homieiot/homie-esp8266 #546: "device_stats_interval": 900, ist not correct shown in mqtt but works.

JorinL, 2018-10-22:
Hey there, I tried to configure Homie that it sends out every 900 seconds the stats.

`{
  "name": "sonoff-s20-1",
  "device_id": "unicolor_rgb_strip_pc",
  "device_stats_interval": 900,
  "wifi": {
    "ssid": "XXXXXXXXXXXXXXX",
    "password": "XXXXXXXXXXXXXX"
  },
  "mqtt": {
    "host": "192.168.0.8",
    "port": 1883,
    "base_topic": "homeland/haushalt/elektrik/rgb/",
    "auth": true,
    "username": "XXXX",
    "password": "XXXXXXXXXXXXXXXXXXXXX"
  },
  "ota": {
    "enabled": true
  }
}`

Optical that leads to - but it is working:
> homeland/haushalt/elektrik/sonoff/sonoff-s20-1/$stats/interval 0

> homeland/haushalt/elektrik/sonoff/sonoff-s20-1/$stats/signal 100
homeland/haushalt/elektrik/sonoff/sonoff-s20-1/$stats/uptime 1803
homeland/haushalt/elektrik/sonoff/sonoff-s20-1/$stats/signal 100
homeland/haushalt/elektrik/sonoff/sonoff-s20-1/$stats/uptime 2703

mwildbolz, 2018-10-27:
it is written as $stats/interval 0 because of the division by 1000, already reported as bug in #521

JorinL, 2018-10-28:
Ah :-) fine!

JorinL, 2018-10-28:
Duplicate of #521

closed 2018-10-28 by JorinL