homieiot/homie-esp8266 #551: /homie/config.json broken when uploading Homie 2.0.0 on top of other sketch

igrowing, 2018-11-08:
This issue is similar to #382 and a little different.

The EspEasy was burnt on the ESP-12 before Homie 2.0.0. 2 boards with the same behavior.
The compiler setting is `build_flags = -Wl,-Teagle.flash.4m.ld`

The Homie reports over serial:
```
✖ /homie/config.json doesn'trl␀l��|␀�l�|␂␌␌␌�␌
```

To fix that I tried following:

- SPIFFS.format(); on boot.
- Remove /homie/config.json.
- Create empty /homie/config.json.
- Create /homie/config.json with all valid fields empty.
- Create /homie/config.json with valid values in valid fields.
- Full erase the Flash and redoing the above.

No success so far.
Burning the same firmware of Homie on new ESP-12 module works as charm. This means the problem isn't in the build.
I think there are some residues of old firmware somewhere beside Flash... Why do I think so?
Because the ESP-12 enters AP mode with broadcasing SSD like EspEasy `ESP_123456`. How this can be if I fully erased the 4MB flash with original Espressif Flashing tool?

Any idea to workaround the problem is welcome.

e1970180, 2019-05-30:
I met with similar issue and find the solution for my case.

**Problem Description**
The config.json was uploaded.
After device power switch off and on I received a log message that config.json doesn't exist and device goes to config mode.
Problem is reproduceable on differen devices D1 mini as well with differen ESP8266 core versions (2.5.2, 2.5.0, 2.4.2) 

**Solution I find**
I learned, that config.json “disappears” from file system, while other uploaded files (e.g. “ui_bundle.gz” and “test.json”) are still there. 
The solution I find is to put a delay in Config::load() in Confi.cpp file:  

```
bool Config::load() {
  if (!_spiffsBegin()) { return false; }

  _valid = false;

 if (!SPIFFS.exists(CONFIG_FILE_PATH)) {
   Interface::get().getLogger() << F("✖ ") << CONFIG_FILE_PATH << F(" doesn't exist") << endl;
   return false;
 }

delay(100);	// <-- this delay 100ms or more helps

  File configFile = SPIFFS.open(CONFIG_FILE_PATH, "r");
  if (!configFile) {
    Interface::get().getLogger() << F("✖ Cannot open config file") << endl;
    return false;
  }
```

@igrowing  Would you please check if this solve your issue too and report?

Best regars, Alex

stritti, 2020-12-28:
No response of @igrowing. Closing issue.

closed 2020-12-28 by stritti