homieiot/homie-esp8266 #553: Hardcoding MQTT Server Credentials

fender21, 2018-11-26:
I would like to hardcode my MQTT Server Credentials. I attempted to do this via the config file but it still prompts the user when joining a network for the MQTT Server/port/user/password? I'd like to keep that hidden, so I'm curious if there is a way to do this?

mchestr, 2018-11-29:
I have done something similar, but don't have code samples at the moment, but it is possible with some modifications.

If you are using the UI bundle, you can build your own without the MQTT page by cloning this repo: https://github.com/marvinroger/homie-esp8266-setup and removing the MQTT portions in App.vue and ProgressBar.vue.

Then you should be able to hardcode the MQTT config right before the validation here: https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Boot/BootConfig.cpp#L394

Something Like:
```
parsedJson["mqtt"]["host"] = MQTT_HOST;
...
```

I haven't tested this, but I have done something similar in a project I am working on. This is just one of the ways to accomplish what you want.

stritti, 2019-04-04:
@fender21 has the workaroud of @mchestr solved your problem? Then please close this issue.

bertmelis, 2019-04-04:
Keep in mind that in order to keep the credentials hidden, you have to encrypt the binary. Although in binary, the strings are hardcoded in plain text...

--> you can just read the hardcoded values via subscribing the the OTA topic.

kleini, 2019-04-04:
Either some configuration or the compiled binary is stored on the flash chip. And even if it would be stored encrypted, the encryption password needs to be stored on the flash, too. So just by reading the source code and the flash chip everybody can decode and read the MQTT server credentials and WiFi credentials. Therefore the whole approach to hardcode or encrypt things does not help anything at all.

You have to ensure, that nobody is able to get hardware access to your Homie devices!

closed 2019-04-04 by kleini