homieiot/homie-esp8266 #557: Issue integrating with OpenHab

marcolino7, 2019-01-08:
Hi,
I'm trying to use Homie ESP with OpenHab MQTT Binding version 2.
To permit OpenHab to mark a Device Online, Device should send the $state node in this format:

homie/deviceID/$state

It seems that Homie for ESP do not send this information to Broker.

OpenHab docs: https://www.openhab.org/addons/bindings/mqtt.generic/

Many Thanks

euphi, 2019-01-08:
homie-esp8266 does not yet follow the 3.0 version of the convention.

However, there is a branch/PR that implements the V3 convention, see https://github.com/bodiroga/homie-esp8266#develop-v3 .

If you use platformio, you can just replace "Homie" with this URL in the lib_deps to test. The signature of the HomieNode-constructor has changed and of the handleInput() callback, so you may need to adapt your sources.

(For an example what to change when moving from homie-esp8266 release to the v3.0-development branch, see https://github.com/euphi/HomieLoggerNode/commit/48a56f27ed4394dcc894d4d76413f98917d699d4 )

marcolino7, 2019-01-10:
Hi,
thanks for help, it seems to work.
Will be homie-esp8266 ported officially for Homie 3.0.0 implementation?

Marco

euphi, 2019-01-22:
I merged it, but it is still a unreleased development version.

--> https://github.com/homieiot/homie-esp8266/tree/develop-v3

So use `https://github.com/homieiot/homie-esp8266#develop-v3` for platformio

closed 2019-01-22 by euphi

marcolino7, 2019-01-22:
Thanka a lot