homieiot/homie-esp8266 #558: Homie wrapper to make examples more useful

lingwooc, 2019-01-16:
When I was setting up my homie based heating system I created a project where I slightly restructured the way the homie code is set up. I found I was repeating myself more that I would expect to and simple things weren't already implemented

It works as a wrapper round the current version of homie at the time (about a year ago). Here's an example for a system with a button and an LED.
```
#include <HomieExtras.hpp>

#define PLUG_RELAY 12
#define BUTTON_PIN 0

DigitalOutNode plugNode("plug", PLUG_RELAY, HIGH);
DigitalInEventNode buttonNode("button", BUTTON_PIN);

void setup()
{
  Serial.begin(115200);
  Serial << endl
         << endl;

  Homie_setFirmware("mqtt-plug", "1.0.0");

  HomieExtras::setup();
  Homie.setup();
}

void loop()
{
  Homie.loop();
}
```
That is all the code needed for an LED to be bound to an MQTT event and a button to send debounced events on press. The same approach can be used for temp sensors. I've started to implement some other bits but didn't get far beyond buttons and temp sensors.

I'll probably keep developing it regardless as I prefer working this way but I thought it might be a good structure for implementations of standard hardware to live. It makes them more usable than the example folder.
Anyway here it is, if it's interesting, good! If not, nothing is lost.
https://github.com/lingwooc/homie2-extras

amayii0, 2019-01-16:
I'm lazy and I like your code :)

euphi, 2019-01-17:
Looks nice, but why do you wrap around HomieNode instead of deriving from it? What is the benefit?

lingwooc, 2019-01-17:
I guess because it keeps the api cleaner, but only mildly. I'm trying to hide the HomieNode in the new abstraction so it seemed natural to hide it properly. Reduces the temptation to break out of the pattern I'm using. I guess it uses a tiny bit more memory but at least in my application that isn't close to being a problem. It would work either way.