homieiot/homie-esp8266 #569: Cannot compile example provided - "shutters"

suprimex, 2019-02-21:
Arduino 1.8.5 on Windows.
All libraries used in sketch updated to latest version ( 21/feb/2019)

Here is an output:

SonoffDualShutters:60: error: no matching function for call to 'Shutters::Shutters(const long unsigned int&, void (&)(), void (&)(), void (&)(), uint8_t (&)(), void (&)(uint8_t), const float&, void (&)(uint8_t))'

 Shutters shutters(COURSE_TIME, shuttersUp, shuttersDown, shuttersHalt, shuttersGetState, shuttersSetState, CALIBRATION_RATIO, onShuttersLevelReached);

                                                                                                                                                     ^

C:\Users\<my-user-name>\Documents\Arduino\libraries\homie-esp8266\examples\SonoffDualShutters\SonoffDualShutters.ino:60:149: note: candidates are:

In file included from C:\Users\<my-user-name>\Documents\Arduino\libraries\homie-esp8266\examples\SonoffDualShutters\SonoffDualShutters.ino:22:0:

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:75:3: note: Shutters::Shutters()

   Shutters();

   ^

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:75:3: note:   candidate expects 0 arguments, 8 provided

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:39:7: note: constexpr Shutters::Shutters(const Shutters&)

 class Shutters {

       ^

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:39:7: note:   candidate expects 1 argument, 8 provided

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:39:7: note: constexpr Shutters::Shutters(Shutters&&)

C:\Users\<my-user-name>\Documents\Arduino\libraries\arduino-shutters\src/Shutters.h:39:7: note:   candidate expects 1 argument, 8 provided

Using library homie-esp8266 at version 2.0.0 in folder: C:\Users\<my-user-name>\Documents\Arduino\libraries\homie-esp8266

gkaradimas, 2019-06-07:
Is there any update in the issue?
The example still doesn't compile.
Do we need an other version of the Shutters library: https://github.com/marvinroger/arduino-shutters ?

codefrey, 2019-08-21:
Is this still an important issue? Should I just fix it?

stritti, 2020-12-28:
Fixed by PR #623

closed 2020-12-28 by stritti

luebbe, 2020-12-28:
propably not fixed, because the PR never got merged