homieiot/homie-esp8266 #571: if MQTT server goes down, the Homie devices stop responding to physical keys

manuelmouraguedes, 2019-03-04:
Hi, 

Just implemented Homie on a Tuya based wall switch (2 buttons). Connects to Mosquitto under HASSIO and everything is working great.
BUT, if I stop the MQTT server the physical buttons stop responding and any rebbot hangs with the repeating  debug message "Attempting to connect to MQTT..."
So, if MQTT is down the switch cannot perform is offline capabilities, i.e react to physical button press and trigger relay
I intend to replace all my wall switches by these Homie/Tuya ones. Cannot have the wall switches dead if the Wifi or Mosquitto goes down...
Is this how Homie works, requiring that MQTT will always be available to respond or is there a confuguration that enables the switch to communicate with MQTT when available but still work offline when MQTT is not reachable?

Thanks,

euphi, 2019-03-07:
The loop handler is not called, if MQTT is disconnected.

This was already discussed earlier. For version 2 there is a PR that has not yet been merged: #490. You can manually add it to your sources.

For version 3 I will provide a PR soon (and also care for tests).

euphi, 2019-03-10:
Closed with #490.

closed 2019-03-10 by euphi