homieiot/homie-esp8266 #573: [develop-v3] Build error in Branch develop-v3

stritti, 2019-03-07:
The current version in branch 'develop-v3' can not be compiled. There is following error:

```
.piolibdeps\Homie\src\Homie\Config.cpp: In member function 'bool HomieInternals::Config::_spiffsBegin()':
.piolibdeps\Homie\src\Homie\Config.cpp:13:37: error: no matching function for call to 'fs::FS::begin(bool)'
_spiffsBegan = SPIFFS.begin(true);
^
```
see: 
https://github.com/homieiot/homie-esp8266/blob/5b34a624950f533bf85b42efad1158bed7823caf/src/Homie/Config.cpp#L13

I used current `espressif8266 @ 2.0.4`  to compile project and used following settings in platformio.ini:
```
platform = espressif8266
board = nodemcuv2
framework = arduino

;Add the PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY build flag to ensure reliable OTA updates.
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY

lib_deps =
  https://github.com/homieiot/homie-esp8266.git#develop-v3
```

stritti, 2019-03-07:
See also at Travis-Build of my example: https://travis-ci.org/stritti/Homie8266-Ping

stritti, 2019-03-07:
I think it is caused by merging ESP32 PR:
https://github.com/bodiroga/homie-esp8266/pull/1#issuecomment-454373132

stritti, 2019-03-08:
PR created.

euphi, 2019-03-09:
PR merged. Many thanks!

closed 2019-03-09 by euphi