homieiot/homie-esp8266 #575: Esp32, OTA, MD5 checksum

jeanvdr, 2019-03-09:
As you all probably know already, the Esp32-OTA implementation is not working properly.
There's a problem with the getSketchMD5 esp32 function which ends in a boot loop. This function works properly in other sketchs and if we move the MD5 calculation from where it is (begining of bootnormal.cpp) in the bootnormal::setup procedure, it works without any problems.

bertmelis, 2019-04-04:
`ESP.getSketchMD5()` is called in the BootNormal constructor. I also suggest to move this into `BootNormal::setup()` for both ESP8266 and ESP32. It's always tricky to call functions/methods in constructors for global objects.

Further improvement is to call `Update.end()` in `onMqttConnected()`. A disconnection during OTA ends up in a stale OTA update. Restarting an update doesn't work because `Update.begin()` returns `false` as there is still an update running.

bodiroga, 2019-04-24:
Fixed by #593 😉

PR #593 (2019-04-24): Fix OTA problems for ESP32
Hi guys!

Thanks to the great instructions provided by @jeanvdr and @bertmelis, OTA updates are now working for ESP32 boards. I don't have any ESP8266 with me right now, so please test it on those boards before merging!

Closes #575
closed 2019-04-24 by bodiroga