homieiot/homie-esp8266 #585: [develop-v3] Build error due to wrong library for ESP32

EinfachArne, 2019-03-17:
Hey,

today I tried to compile the current development version of Homie-V3 via PlatformIO fpr an ESP32 and got the following error:

```
Compiling .pioenvs\lolin32\src\window_sensor.cpp.o
In file included from .piolibdeps/Homie/src/Homie.hpp:12:0,
from .piolibdeps/Homie/src/Homie.h:4,
from src\window_sensor.cpp:1:
.piolibdeps/Homie/src/Homie/Boot/BootNormal.hpp:21:25: fatal error: ESP8266mDNS.h: No such file or directory

*********************************************************************
* Looking for ESP8266mDNS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266mDNS.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266mDNS.h
*
*********************************************************************

compilation terminated.
Compiling .pioenvs\lolin32\libaab\ESP Async WebServer_ID306\AsyncEventSource.cpp.o
Compiling .pioenvs\lolin32\libaab\ESP Async WebServer_ID306\AsyncWebSocket.cpp.o
*** [.pioenvs\lolin32\src\window_sensor.cpp.o] Error 1
Compiling .pioenvs\lolin32\libaab\ESP Async WebServer_ID306\SPIFFSEditor.cpp.o
```

My platformio.ini looks like that:

```
[env:lolin32]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = lolin32
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
monitor_speed = 115200
lib_deps = https://github.com/homieiot/homie-esp8266.git#develop-v3
```

The reason for that seems to be the use of a wrong library in scr/Homie/Boot/BootNormal.hpp. Instead of "ESP8266mdns.h" it needs to be "mdns.h" since only that is included in arduinoespressif32. After I changed it I was able to compile.

cobii, 2019-03-17:
Have had the same issue. Your solution works also in Hamburg. Danke!

euphi, 2019-03-18:
Sorry for that. Seems that I wasn't careful enough when merging the "disabling MDNS" commit from v2-#develop.

commit a9932245 (2019-03-18): Fix ESP32 compilation (wrong include for MDNS)

Fix #585
closed 2019-03-18 by euphi