homieiot/homie-esp8266 #588: [develop-v3] Core panics after connection to AP

EinfachArne, 2019-03-20:
Hey,

everything works fine until I connect to my ESP32 via WiFi. After that I keep getting the following error message in an endless loop:

```
✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware bare-minimum (1.0.0)
🔌 Booting into config mode 🔌
Device ID is 30aea41f1b18
AP started as Homie-30aea41f1b18 with IP 192.168.123.1
Triggering Wi-Fi scan...
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8010ba31  A1      : 0x3ffb4e20
A2      : 0x3ffd03a4  A3      : 0x3ffcfde0  A4      : 0x3ffce900  A5      : 0x3ffce654
A6      : 0x027ba8c0  A7      : 0x0c7ba8c0  A8      : 0x8010b8d0  A9      : 0x3ffb4de0
A10     : 0x3ffd03b4  A11     : 0x3ffcfde0  A12     : 0x3ffb4e2c  A13     : 0x00000044
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x00000000:0x3ffb4e20 0x4010ba2e:0x3ffb4e60 0x40101e21:0x3ffb4e80 0x401057f5:0x3ffb4ec0 0x4010aa72:0x3ffb4ee0 0x400fa53b:0x3ffb4f00 0x4009180d:0x3ffb4f30

Rebooting...
```

After I disconnect the core stops to panic but that makes it impossible to upload a config.

I used the current development version of Homie-V3 on a Wemos Lolin32 V1.0.0.

Any ideas what the cause could be?

bertmelis, 2019-03-21:
Could you decode the backtrace?

euphi, 2019-03-21:
I don't use config mode, so I did not tested it yet. 

What happesn, if you upload a `homie/config.json` to SPIFFS?

EinfachArne, 2019-03-21:
@euphi Works like a charm and solves at least my problem for now. But of course not the problem in general.

@bertmelis I am not that skilled. Is there a simple way to do it? I am working with PlatformIO on Windows. A quick research brought nothing up.

stritti, 2019-03-23:
@EinfachArne Yes this is a big gap in PlatformIO. There is still missing decoding of back traces.
My current workaround: 
* Start Arduino IDE with installed https://github.com/me-no-dev/EspExceptionDecoder
* open decoder from menu (with no selected/compiled) project)
* the decoder asks for the firmware.bin. You should select the version PlatformIO has built within project sub-direcory `.pioevs/`
* now you could paste the back trace to the opened window

Hope this helps.

bertmelis, 2019-03-23:
You could also do it manually (or create a Windows batch file):

```
%homedrive%%homepath%/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-addr2line.exe -fp -e ./.pioenvs/esp8266/firmware.elf ...
```
if you run `xtensa-lx106-elf-addr2line.exe -h` you can see the how it's used.

EinfachArne, 2019-03-23:
Thank you guys! I think I got it. :)

Ok, for the record:

Tested on Wemos Lolin32 V1.0.0 and ESP32 DevModule - both show the same result. This is based on the DevModule:

```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8010ba31  A1      : 0x3ffb4e20
A2      : 0x3ffcf58c  A3      : 0x3ffcfd84  A4      : 0x3ffcc214  A5      : 0x3ffcc158
A6      : 0x027ba8c0  A7      : 0x0c7ba8c0  A8      : 0x8010b8d0  A9      : 0x3ffb4de0
A10     : 0x3ffcf59c  A11     : 0x3ffcfd84  A12     : 0x3ffb4e2c  A13     : 0x00000044
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x00000000:0x3ffb4e20 0x4010ba2e:0x3ffb4e60 0x40101e21:0x3ffb4e80 0x401057f5:0x3ffb4ec0 0x4010aa72:0x3ffb4ee0 0x400fa53b:0x3ffb4f00 0x4009180d:0x3ffb4f30
```

Decoded:
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x4010ba2e: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x40101e21: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x401057f5: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x4010aa72: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x400fa53b: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x4009180d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

euphi, 2019-03-24:
Oh.. doesn't look good. Maybe a heap corruption?

nemiliani, 2019-07-08:
Hi all, I am experiencing the same thing using an espduino-32, platformio and ubuntu 18.04.

```
Device ID is 3c71bf15651c
AP started as Homie-3c71bf15651c with IP 192.168.123.1
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x801017a1  A1      : 0x3ffb57f0  
A2      : 0x3ffd0948  A3      : 0x3ffd0b94  A4      : 0x3ffd0b7c  A5      : 0x3ffd0164  
A6      : 0x027ba8c0  A7      : 0x0c7ba8c0  A8      : 0x80101640  A9      : 0x3ffb57b0  
A10     : 0x3ffd0958  A11     : 0x3ffd0b94  A12     : 0x3ffb57fc  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb57f0 0x4010179e:0x3ffb5830 0x4010d2b1:0x3ffb5850 0x40111ab9:0x3ffb5890 0x40116d36:0x3ffb58b0 0x40104703:0x3ffb58d0 0x40088fcd:0x3ffb5900
```

Demangled :
```
$ /home/nemi/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-addr2line --exe=.pioenvs/espduino32/firmware.elf -C 0x00000000:0x3ffb57f0 0x4010179e:0x3ffb5830 0x4010d2b1:0x3ffb5850 0x40111ab9:0x3ffb5890 0x40116d36:0x3ffb58b0 0x40104703:0x3ffb58d0 0x40088fcd:0x3ffb5900
??:0
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c:1031
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c:401
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:740
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:184
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:483
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```

I did go to 
```
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!
```
and verified the ssid string at that point is empty, not sure if that is causing the later crash. I will try to debug it and see if I can get to the bottom. If you have any suggestions on what to look for please let me know.

Thanks.

reapzor, 2019-08-04:
Summary:
I reproduced this in the Arduino IDE with just the WiFi library. This is not a Homie bug, it is some bug with the WiFi library and esp32, and setting up a STA_AP where the SSID has the mac address of the device at the end of the string. Any SSID that does not have the mac address of the ESP32 as the last character set in the string seems to work just fine.

Data:
The problem is with the DeviceID string (Mac address).
Not the string itself, but the actual mac address that is returned.

I cannot fathom the issue, but there is only one way I have been able to induce this bug, and it is to host the AP with the device ID at the end of the string. 

Ways to work around this:
-Append addtional text to the SSID so that the mac address is not at the end of the string
--"Homie-d8a01d54722d-ap" - Works fine, no crashing when hosting AP
--"Homie-d8a01d54722dZZ" - Also works fine, no crashing
- Dont use mac address in the SSID at all
--"Homie" Works fine, no crashing

Ways to create this error/bug:
-Have mac address as the last bit of the SSID in any given SSID
--"AnythingHered8a01d54722d" Crashes
--"AnythingAtAll-d8a01d54722d" Crashes

I reproduced this in the Arduino IDE with just the WiFi library. This is not a Homie bug, it is some bug with the WiFi library and esp32, and setting up a STA_AP where the SSID has the mac address as the end of the string.

The best workaround for this would be to append "-ap" or some other text to the end of the AP name in BootConfig.cpp so that the SSID does not end with the mac address of the device.

Example working solution in BootConfig.cpp
```
  char apName[MAX_WIFI_SSID_LENGTH];
  strlcpy(apName, Interface::get().brand, MAX_WIFI_SSID_LENGTH - 1 - MAX_MAC_STRING_LENGTH);
  strcat_P(apName, PSTR("-"));
  strcat(apName, DeviceId::get());
  strcat(apName, PSTR("-ap"));
```

Also weird, is you have to upload the sketch twice for the crash to start happening.

codefrey, 2019-08-22:
I created a pull request for a workaround of this issue. Disabling WiFi persistence will get rid of this problem, but may slightly increase the WiFi connection time.

The memory (or at least the part where the WiFi persistence data is stored) needs to be wiped beforehand if a firmware with WiFi persistence enabled has already been used prior.