homieiot/homie-esp8266 #589: [develop-v3] ota_updater not working, missing $online topic

BasvanH, 2019-03-26:
Hello,

I'm trying to ota to a develop-v3 running esp8266. The ota_updater script is stuck on "Waiting for device to come online...". Looking at the code it's waiting for the $online topic, which I do not see coming from the home v3 device.

I'm certain I have the correct base topic & device id as I see all kinds of other messages coming by, just not the $online.

ota is enabled via json config. Confimed with $implementation/config .

Regards,
Bastiaan

euphi, 2019-03-26:
I changed the $online topic to $state, see 83cf3acca108ade2a741507f264d2d18691758f9

Now OTA works, however the python scripts seems to compare the old checksum instead of the correct new one?!

Furthermore, the esp8266 crashes after OTA, but comes up again after WDT timeout with correct new firmware.

BasvanH, 2019-03-27:
Using the $state topic works. I do not have the problems with the checksum, it validates correctly.

euphi, 2019-03-27:
It seems that the update scripts receives the old checksum again after the `$state` changes to `init`, altough the esp only sends the new checksum after restart. (checked with mosquitto_sub).

There seems to be a retained message with the old checksum, however, I do not understand why the updater-script receives it again.
I can see that the same issue happens with the firmware version.

bertmelis, 2019-04-18:
> It seems that the update scripts receives the old checksum again after the `$state` changes to `init`, altough the esp only sends the new checksum after restart. (checked with mosquitto_sub).

Same here. Using Python 2.7.13
Although the device has updated successfully, the script indicates it has failed.

euphi, 2020-01-09:
Oh. I don't remember this issue, although I even changed something on the script for it...
Two important remarks:

* The update script shall allow both `$online` or `$state` for starting OTA - so it also possible to use OTA for update from 2.0 to 3.x

* The "wrong checksum" bug is still present, so it needs to be fixed.

commit b9f6bc19 (2020-01-10): Fix #589

One line to fix both problems mentioned in #589:

Checking the "$state" topic for not "false" was wrong, because since v3
the online state must be "ready" to continue.

Thus, also in "init" state the ota-updater re-subscribed to the $fw
topic and thus got old retained checksum from broker. (The homie device
usually has not sent the new $fw/checksum at this moment).

Now the script uses the firmware MD5 only if it has been sent from a
"ready" device (or $online != false device to support v2.x devices).
closed 2020-01-10 by euphi