homieiot/homie-esp8266 #596: Wiki: make editable for everybody

stritti, 2019-04-17:
I think Wiki should be maintainable for everybody to get more flexibility and be more close to community.

kleini, 2019-04-17:
Is it possible to integrate a review by contributors before changes are published? There are maybe some trolls out there, trying to destroy things, although I hope, this will not be the case.

stritti, 2019-04-17:
You need at least a GitHub account. So not really everyone... I think reviews are not required if I have a look at other projects.

luebbe, 2019-04-18:
according to the gitter chat there was only one person interested in editing wiki pages. Why don't we just make this person a contributor?

stritti, 2019-04-18:
Hm, I think it could be another level to contribute without being contributor.

stritti, 2019-04-23:
Close issue: less feedback, would not be changed.

closed 2019-04-23 by stritti