homieiot/homie-esp8266 #601: MQTT connection is lost if SSL is enabled and too much data is transferred

kleini, 2019-05-02:
I try to run some ESP8266-12E now with three Homie nodes:
- DS18B20Node from Homie node collection
- DHT22Node from Homie node collection
- AdcNode from Homie node collection

The MQTT connection is SSL encrypted using #577.

After adding the AdcNode the MQTT connection is immediately lost after AdcNode tries to sent its properties. If I remove the AdcNode completely from the code, everything works without any problems.

💡 Firmware esp8266-12e (0.6.4)
🔌 Booting into normal mode 🔌
SSL is: 1
{} Stored configuration
  • Hardware device ID: cc50e309ea0b
  • Device ID: esp8266-12e
  • Name: ESP8266 12E
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: MySSID
    ◦ Password not shown
    ◦ IP: 192.168.x.x
    ◦ Mask: 255.255.255.0
    ◦ Gateway: 192.168.x.1
  • MQTT: 
    ◦ Host: 192.168.x.y
    ◦ Port: 1883
    ◦ SSL enabled: true
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: esp8266-12e
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
  • Custom settings: 
    ◦ adcCorrect: 0.97 (set)
    ◦ battMax: 3.30 (default)
    ◦ battMin: 2.60 (default)
• DS18B20 sensor pin[4]:
  ◦ Reading interval: 60 s
  ◦ Found 1 sensors.
• DHT22 sensor pin[13]:
  ◦ Reading interval: 60 s
• ADC measurement:
  ◦ Send interval: 300 s
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.x.x
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 5s
• DS18B20 sensor pin[4]:
  ◦ Temperature: 19.31 °C
• DHT22 sensor pin[13]:
  ◦ Temperature: 20.10 °C
  ◦ Humidity: 60.60 %
• ADC measurement:
  ◦ Voltage: 3.29V
  ◦ Battery level: 98.33%
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
MQTT disconnected, reason: 0
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 7s
Prepare to sleep.
Flagged for sleep by sketch
Device in preparation to sleep...
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
MQTT disconnected, reason: 0
Triggering READY_TO_SLEEP event...
💤 Device is deep sleeping...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...

kleini, 2019-05-02:
Without AdcNode everything works and MQTT connection is not lost.

💡 Firmware esp8266-12e (0.6.5)
🔌 Booting into normal mode 🔌
SSL is: 1
{} Stored configuration
  • Hardware device ID: cc50e309ea0b
  • Device ID: esp8266-12e
  • Name: ESP8266 12E
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: xxxx
    ◦ Password not shown
    ◦ IP: 192.168.x.y
    ◦ Mask: 255.255.255.0
    ◦ Gateway: 192.168.x.1
  • MQTT: 
    ◦ Host: 192.168.x.z
    ◦ Port: 1883
    ◦ SSL enabled: true
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: esp8266-12e
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
• DS18B20 sensor pin[4]:
  ◦ Reading interval: 60 s
  ◦ Found 1 sensors.
• DHT22 sensor pin[13]:
  ◦ Reading interval: 60 s
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.x.y
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 45s
• DS18B20 sensor pin[4]:
  ◦ Temperature: 19.37 °C
• DHT22 sensor pin[13]:
  ◦ Temperature: 20.10 °C
  ◦ Humidity: 63.70 %
Prepare to sleep.
Flagged for sleep by sketch
Device in preparation to sleep...
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
MQTT disconnected, reason: 0
Triggering READY_TO_SLEEP event...
💤 Device is deep sleeping...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...

kleini, 2019-05-05:
This issue seems to be cause by the implementation of AsyncMQTT and not by Homie. Need to open my issue there.

closed 2019-05-05 by kleini

luebbe, 2019-05-06:
I have a similar issue when a device with several nodes (in my case a BME80 + DHT22) connects. A part of the information which the nodes try to send upon MQTT connect is lost. In your case it even kills the connection itself.

kleini, 2019-05-06:
I am glad, that I am not the only one. @luebbe do you use SSL encryption? I wanted to test this now without encryption.

luebbe, 2019-05-07:
No, I'm not using SSL.

kleini, 2019-05-07:
Interesting. I thought it just happens due to SSL problems with too few memory. But if this happens without using SSL encrypted connections, it must be some general problem with Async MQTT.

luebbe, 2019-05-07:
Maybe related to https://github.com/homieiot/homie-esp8266/issues/515? I commented there

kleini, 2019-05-07:
Sounds like that.