homieiot/homie-esp8266 #609: v2 dht22 sensor disconnects after several hours

bmbouter, 2019-06-25:
First off I really like using your implementation of Homie for the esp8266. I've ordered some ESP32 modules to try your 3.0 version at some point. For now though I've got an issue with my nodemcu working great for a few hours, but then disconnecting.

# It works

This will work great for me for hours:

```
#include <Homie.h>

#include <Adafruit_Sensor.h>
#include <DHT.h>

const int TEMPERATURE_INTERVAL = 5;

/**************************** PIN DEFINITIONS *********************************/
#define PIRPIN    D5
#define DHTPIN    D7
#define DHTTYPE   DHT22
#define LDRPIN    A0

int pirValue;
int LDR;

// DHT_Unified dht(DHTPIN, DHTTYPE);
DHT dht(DHTPIN, DHTTYPE);

unsigned long lastTemperatureSent = 0;

HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");
HomieNode motionNode("motion", "motion");
HomieNode lightNode("light", "light");
HomieNode ledNode("led", "led");

void setupHandler() {
  temperatureNode.setProperty("unit").send("°F");
  humidityNode.setProperty("unit").send("%");
  motionNode.setProperty("unit").send("motion");
  lightNode.setProperty("unit").send("light");
}

unsigned long start, finished, elapsed;

void loopHandler() {
  if (millis() - lastTemperatureSent >= TEMPERATURE_INTERVAL * 1000UL || lastTemperatureSent == 0) {

    start=millis();

    // Reading temperature or humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
    float h = dht.readHumidity();
    // Read temperature as Fahrenheit (isFahrenheit = true)
    float t = dht.readTemperature(true);

    // Check if any reads failed and exit early (to try again).
    if (isnan(h) || isnan(t)) {
      Homie.getLogger() << "Failed to read from DHT sensor!" << endl;
      return;
    }

    Homie.getLogger() << "Humidity: " << h << " %" << endl;
    humidityNode.setProperty("percentage").send(String(h));

    Homie.getLogger() << "Temperature: " << t << " °F" << endl;
    temperatureNode.setProperty("degrees").send(String(t));

    pirValue = digitalRead(PIRPIN); //read state of the PIR
    Homie.getLogger() << pirValue << endl;

    if (pirValue == LOW) {
      motionNode.setProperty("motion").send(String(0));
    }

    else if (pirValue == HIGH) {
      motionNode.setProperty("motion").send(String(1));
      Homie.getLogger() << "Motion Detected" << endl;
    }

    LDR = analogRead(LDRPIN);
    Homie.getLogger() << "light (0 - 1024): " << LDR << endl;
    lightNode.setProperty("light").send(String(LDR));

    lastTemperatureSent = millis();

    finished=millis();
    elapsed=finished-start;
    Homie.getLogger() << "loop Runtime: " << elapsed << endl;
    Homie.getLogger() << "start: " << start << endl;
    Homie.getLogger() << "finished: " << finished << endl << endl;
  }
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;

  Homie_setFirmware("multi-sensor", "1.0.0");
  Homie.setSetupFunction(setupHandler).setLoopFunction(loopHandler);

  pinMode(PIRPIN, INPUT);
  pinMode(DHTPIN, INPUT);
  pinMode(LDRPIN, INPUT);

  dht.begin();

  temperatureNode.advertise("unit");
  temperatureNode.advertise("degrees");

  humidityNode.advertise("unit");
  humidityNode.advertise("percentage");

  motionNode.advertise("unit");
  motionNode.advertise("motion");

  Homie.setup();
}

void loop() {
  Homie.loop();
}

```

# What breaks?

After some random number of hours when I walk back to the nodemcu the light is blinking with a fast blink like it's waiting to connect to MQTT and somehow it's lost its connection.

# Observations

My main concern was that the `loopHandler()` was executing for longer than the recommended 50 ms, but after adding some timings when not reading the sensors it runs in ~ 1 ms and when using the sensors ~10-12 ms max so that is much less than 50.

# Possible Issues?

What if ocassionally `loopHandler()` does take a really long time? What would that do?

# Not a hardware issue (or a really exceptional one) and strangeness

My friend also runs another nodemcu w/ the same code and experiences the same issue (unreliability after several hours). Very strangely even while Homie is fast-blinking he can still issue commands to his setabble property which controls an LED, so that is extremely strange.

# Interrupts?

Is the `loopHandler()` configured to be interrupted by the wifi stack?

n8twj, 2019-06-25:
Hi - check your mqtt broker for any error messages.

I am experiencing  perhaps a similar situation. My MQTT broker complains
about an invalid QOS value in subscription. I’m still investigating.

stritti, 2020-01-03:
Could you please shift to v3? If still having ploblems, please reopen issue then.

closed 2020-01-03 by stritti