homieiot/homie-esp8266 #610: MQTT wont auto-reconnect

dakhnod, 2019-07-12:
When my ESP loses connection to MQTT an event is triggered, and the ESP tries to reconnect exactly once.

If My Server is not up for that try, the ESP does not seem to retry connecting to MQTT.

I think the reason is the mqttReconnectTimer [not being reactivated](https://github.com/homieiot/homie-esp8266/blob/develop-v3/src/Homie/Boot/BootNormal.cpp#L765).

Could anyone explain to me why the _mqttDisconnectNotified flag is relevant here?

Why can't the reconnectTimer be started regardless?

Is a failed MQTT reconnect supposed to trigger _onMqttDisconnected again, leading to the timer being restarted?

stritti, 2019-07-23:
#611

stritti, 2019-08-03:
PR #611 merged and cherry picked to `develop-v3`

closed 2019-08-03 by stritti