homieiot/homie-esp8266 #612: STM32F4  esp8266 OTA using MQTT

Akashmerasplugins, 2019-07-16:
Hi everyone. I am trying to make the OTA update over mqtt but I don't know how to do it. I am using a stm32f429i controller with esp8266 wifi module, I am using cloud mqtt to publish and receive the messages all is working, the only thing is the updates. I am really lost with that. Maybe a tutorial or something like that? I don't know how to receive the file from mqtt and flash it inside the ROM can anyone help me how to do it.
Thanks in advance.
Regards.
Akash

MajorTwip, 2019-07-16:
Is your device allready running a Homie-Sketch?
Have you allready read the docs? 
https://homieiot.github.io/homie-esp8266/docs/develop/others/ota-configuration-updates/

Akashmerasplugins, 2019-07-17:
No, it is not running a Homie-Sketch and I don't know homie.
while collecting information about OTA for stm32 I found that it can be made using IAP(In-Application Programming). By reading about these things in the web I can get to know  how the OTA is done and programmed in flash but, I don't know how to get the data from MQTT and flash it inside the ROM

kleini, 2019-07-17:
@Akashmerasplugins Then this issue tracker is the wrong space. This issue tracker is for the Homie implementation on ESP8266 and ESP32 devices.

closed 2019-07-17 by kleini