homieiot/homie-esp8266 #613: [develop-v3] OTA randomly nearly working, else crash

MajorTwip, 2019-07-16:
Hi all

Trying to do OTA-Update, but with no success.
Approximately 4 out of 5 times i got following Log:

Receiving OTA payload
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary

Exception (3):
epc1=0x40238a90 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40013031 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffea00 end: 3fffffb0 offset: 01a0
3fffeba0:  00000000 00000000 4bc6a7f0 00000000
3fffebb0:  00000000 4020a777 4010027c 00010ae6
3fffebc0:  00000000 3ffee9b8 00000003 40216123
3fffebd0:  013499f0 00000000 3fffec60 00000003
........ [etc]

Which translates to:

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40238a90
EXCVADDR: 0x40013031

Decoding stack results
0x4021ae0b: strtok at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/string/strtok.c line 102
0x4021ae0b: strtok at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/string/strtok.c line 102
0x4021f5f8: operator new[](unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_opv.cc line 33
0x4021f626: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4021f626: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x40223074: tcp_input at core/tcp_in.c line 501
0x40220000: tcp_seg_free at core/tcp.c line 1632
0x40227d59: ip4_input at core/ipv4/ip4.c line 710
0x4021fd49: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x4021fb7f: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 441
0x4024454a: ethernet_input at glue-esp/lwip-esp.c line 363
0x4024455b: ethernet_input at glue-esp/lwip-esp.c line 371

The sketch couldn't be easier:
```
`#include <Arduino.h>
#include <Homie.h>

void setup() {
  Serial.begin(9600);

  // put your setup code here, to run once:
  Homie_setFirmware("csh_primer","0.0.9");
  Homie_setBrand("Comstock Homecontrol");
  Homie.setup();
}

void loop() {
  // put your main code here, to run repeatedly:
      Homie.loop();

}`
```

and fallowing platformio.ini

```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =    
    ArduinoJson@5.13.4
    https://github.com/homieiot/homie-esp8266#develop-v3
build_flags = 
    -D HOMIE_CONFIG=0
    -D HOMIE_MDNS=0
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -Wl,-Teagle.flash.4m2m.ld
```

Following Libs:
```
Dependency Graph
|-- <ArduinoJson> 5.13.4
|-- <Homie> 3.0.0 #1e157d4
|   |-- <ArduinoJson> 5.13.4
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <Bounce2> 2.52
|   |-- <ESP Async WebServer> 1.2.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 5.13.4
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Ticker> 1.0
|   |-- <ESP8266WiFi> 1.0
```
(WebServer and mDNS still in dependencies, even with the Flags in the .ini, but why...)

The Device is in this case a generic NodeMCU (Following output from the uploader)
```
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:7b:2d:65
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 361552 bytes to 254134...
Wrote 361552 bytes (254134 compressed) at 0x00000000 in 22.6 seconds (effective 127.7 kbit/s)...
Hash of data verified.
```

Then it reboots normally with the old FW.

And 1 out of 5 times (same sketch) it loads to approximately 350kb, then MQTT disconnects.

I uses mosquitto and now verneMQ, same behaviour. 
I used the updater.py from the repo (develop-v3) and a custom project https://github.com/MajorTwip/HomieControl
Both with the same result.

MajorTwip, 2019-07-16:
Downgraded just to test to esp8266 android platform 2.0.2, now it works... Will try to find out with which Version it breaks...

MajorTwip, 2019-08-06:
Update; Cleaned up my Library-Chaos and now it is nearly working. No more crashes.
But at approximately 350kb out of 412 there is a MQTT disconnect reason 0.
Yesterday evening, after 2200, when the backups kicked in and the network slowed down, the Mqtt-disconnected appeared allready at 200kb out of 412...
Now I figured out, that the disconnect happens after 40 seconds, not necessarily after 350kb.

With a simpler sketch under 200kb all works fine.

bertmelis, 2019-08-06:
Could you try raising the keepalive for the MQTT connection?
I have had mixed results when the ESP tries to send a ping during OTA (of is it the broker that disconnected?).

MajorTwip, 2019-08-07:
@bertmelis thank you so much.
```
  Homie.getMqttClient().setKeepAlive(120);
  Homie_setFirmware("csh_primer", "0.0.13");
  Homie_setBrand("Comstock Homecontrol");
  Homie.setup();
```
It seems to work now.

closed 2019-08-07 by MajorTwip

MajorTwip, 2020-06-17:
I think I found another solution/source of the problem. If you setup serial with 9600 Baud it slows the update down, so much so that it aborts the OTA. So either disable logging or keep the Baudrate higher.