homieiot/homie-esp8266 #614: Programming a sonoff dual R2 : not working anymore for me..

mosseb, 2019-07-16:
Hi.
With my friend we built our own firmware for sonoff dual r2 using homie.
It has been working for a lot of sonoffs, then I needed to program a new one recently and could not make it work. There has been no change in source code since.
I'm not sure if it's atom related (new version?) or platformio related (new version?) or hardware related (new version of the sonoff ?).
Anyway, I'm not really sure where to post for help because I don't think it's homie related.
Could you point out for some useful resources like forums where to get help ?
Thank you.

amayii0, 2019-07-16:
Hi,
First thing, please define "not working" and also if anything already "works".

mosseb, 2019-07-17:
Erm.. While I started again reproducing the issue for exmplanation, this suddenly fell to working state..
Well sorry for the annoyance...
I suppose I must be happy now :)
Thanks anyway !

closed 2019-07-17 by mosseb