homieiot/homie-esp8266 #615: Does not compile with ArduinoJson 6.x

MajorTwip, 2019-07-17:
Due to breaking changes in the ArduinoJson Library, you need the Version prior to 6.x

kleini, 2019-07-17:
https://github.com/homieiot/homie-esp8266/blob/develop/library.json#L25
The dependency is clearly specified.

closed 2019-07-17 by kleini

MajorTwip, 2019-07-18:
right... interesting, when selecting Update Libraries in PIO it updates never the less to V6... But this is not a problem for this Issue tracker.

kleini, 2019-07-18:
This may be true, that updating libraries updates ArduinoJson in your PlatformIO home directory to version 6.x. How do you compile some sketch using homie-esp8266? In my case PlatformIO compiles my sketches using homie-esp8266 with the correct version of ArduinoJson.

See e.g. here, how I setup my sketches using Homie: https://github.com/kleini/temp-esp8266/tree/develop-v3

MajorTwip, 2019-07-18:
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =    
    ArduinoJson@5.13.4
    https://github.com/homieiot/homie-esp8266#develop-v3
build_flags = 
    -D HOMIE_CONFIG=0
    -D HOMIE_MDNS=0
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -Wl,-Teagle.flash.4m2m.ld

kleini, 2019-07-18:
I used your contents in platformio.ini and added in the src folder a main.cpp with contents from https://homieiot.github.io/homie-esp8266/docs/develop-v3/quickstart/getting-started/#bare-minimum-sketch.
This compiles without any problems for me.

[build.log](https://github.com/homieiot/homie-esp8266/files/3408228/build.log)

So, I still can not reproduce your problem. I am sorry.

iluminat23, 2019-07-19:
I have the same issue as @MajorTwip. I took this as reason to update Homie-ESP8266  to ArduinoJson 6. You can find my work here: https://github.com/iluminat23/homie-esp8266/tree/ArduinoJson6

Currently my changes are only compile time tested.

kleini, 2019-07-19:
@iluminat23 I think, you should create a pull request for your changes. This makes it easier for maintainers to review your changes and furthermore to discuss and improve them.

iluminat23, 2019-07-22:
Yes of course. I just wanted you to know that I'm working on this. So the work is not done twice. After I ran some tests on my h/w I will make a pull request.

iluminat23, 2019-07-23:
After fixing some bugs I have created a pull request #616.