homieiot/homie-esp8266 #621: Weird disconnection issue, uptime, wireless continues, reed switch property stops

psyciknz, 2019-08-23:
Not specifically a homie issue, but I'm using homie and so hopefully there's an error routine someone can help me with (ie I'll detect an issue, but what in the homie framework would i call to continue on).

I've got a reed switch and dht-shield on a d1 mini for my garage door.  I don't know under what condition it is (no real way of debugging OTA is there?) but the uptime and wireless signals will keep posting, but it won't push out an open or closed when the door state changes.

Here's all the code.  
```
#include <Homie.h>
#include <WEMOS_DHT12.h>

#define FW_NAME "dhtshield-reed-dht12"
#define FW_VERSION  "1.0.4"

/* Reed pin */
#define REED_PIN D2
int doorState = 0;

const int TEMPERATURE_INTERVAL = 300;
unsigned long lastTemperatureSent = 0;
DHT12 dht12;

/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */

HomieNode reedNode("reed", "state");
HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");

void setupHandler() {
  // Do what you want to prepare your sensor
  reedNode.setProperty("state").setRetained(true).send("closed");
  pinMode(REED_PIN, INPUT_PULLUP);
  temperatureNode.setProperty("unit").send("c");
  humidityNode.setProperty("unit").send("%");
}

void loopHandler() {
  int lastDoorState = doorState;
  doorState = digitalRead(REED_PIN);
  String doorStateString = "unk";

  if (lastDoorState != doorState) {
    if (doorState == 1) {
      doorStateString = "open";
      //reedNode.setProperty("state").send("open");  
      Homie.getLogger() << "Door State: OPEN" << endl;
    } else if (doorState == 0){
      doorStateString = "closed";
      //reedNode.setProperty("state").send("closed");
      Homie.getLogger() << "Door State: CLOSED" << endl;
    } else {
      //reedNode.setProperty("state").send("unk"); 
      doorStateString = "unk"; 
      Homie.getLogger() << "Door State: UNKNOWN" << endl;
    }
    reedNode.setProperty("state").setRetained(true).send(doorStateString);  
  }

  if (millis() - lastTemperatureSent >= TEMPERATURE_INTERVAL * 1000UL || lastTemperatureSent == 0) {
    if(dht12.get()==0){
        float temperature = dht12.cTemp;
        float humidity = dht12.humidity;
        Homie.getLogger() << "Temperature: " << temperature << " °C" << endl;
        Homie.getLogger() << "Humidity: " << humidity << " %" << endl;
        temperatureNode.setProperty("degrees").send(String(temperature));
        humidityNode.setProperty("percent").send(String(humidity));
        reedNode.setProperty("state").send(doorStateString);  
    }
    
    
    lastTemperatureSent = millis();
  }
  
  delay(1000);
  
  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial << endl << endl;
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie.setSetupFunction(setupHandler).setLoopFunction(loopHandler);

  reedNode.advertise("state");
  temperatureNode.advertise("unit");
  temperatureNode.advertise("degrees");
  humidityNode.advertise("unit");
  humidityNode.advertise("percent");

  Homie.setup();
}

void loop() {
  // put your main code here, to run repeatedly:
  Homie.loop();

}
```

I wonder if the loop handler is getting an error and ending?  And the the homie loop is continuing (to give out the uptime etc).

If I was trapping an error in the loop handler, how will I just accept it and continue on?

It also could be that dht shield causing issues as it doesn't update temps for days.

MajorTwip, 2019-08-24:
Try to eliminate the delay(1000) at the end of the loophandler, probably by making a new if(millis()... like for the dht.

psyciknz, 2019-08-26:
I tried that.  I suspect it's getting an mqtt disconnect, or the wireless might be going....but it's not reconnecting the nodes.

I pushed a just reed switch firmware to it, so no temp readings, and dropped that delay to 100ms.  But I've had the same thing happen again.  

Any other ideas or things i can try?

bertmelis, 2019-08-26:
Why would you need the delay in the first place? Just set it to 1.

Next: is your DHT library blocking? From the code I suspect to. Try to find a non blocking DHT library (perhaps interupt-driven).

psyciknz, 2019-08-26:
So I've removed the dht from the code.  Not sure why the delay was in there.  It always was, possibly because of the dht??  I can remove....Anyway I've managed to log the serial connection on it using a raspberry pi......and I've seen a wifi disconnection and reconnect.  So i then opened the garage door and can see it complaining about setNodeProperty(): impossible now.

```✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 7558s
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 7558s
✔ Wi-Fi connected, IP: 192.168.100.21
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
〽 Sending statistics...
  • Wi-Fi signal quality: 66%
  • Uptime: 7618s
〽 Sending statistics...
  • Wi-Fi signal quality: 66%
  • Uptime: 7678s
〽 Sending statistics...
  • Wi-Fi signal quality: 66%
  • Uptime: 7738s
〽 Sending statistics...
  • Wi-Fi signal quality: 66%
  • Uptime: 7798s
〽 Sending statistics...
  • Wi-Fi signal quality: 66%
  • Uptime: 7858s
✖ setNodeProperty(): impossible now
〽 Sending statistics...
  • Wi-Fi signal quality: 60%
  • Uptime: 7918s
✖ setNodeProperty(): impossible now
```
So need to trace what's happening here.  

It looks like it's connecting to mqtt - hmm but when I look at the mosquitto logs I don't see it connecting.

psyciknz, 2019-08-26:
So if I have a wifi disconnect an dreconnect, what code is run (bu homie) on reconnection?  Does the loopHandler carry on, or does it call setup again?

I see up the nodeProperty right at the top, before being in any functions.  Might this be a null object?  This code has worked for over a year...but I just recompiled it with platformio rather than arduino ide and seeing this problem.

psyciknz, 2019-08-26:
So looking at my installed libraries I see:
repository: https://github.com/.marvinroger/homie-esp8266.git
version 2.0.0

How can I force it to upgrade to a later one?  I think I just installed via teh platformio registry....maybe I need later.

bertmelis, 2019-08-26:
Set your dependency via direct github link and use the develop-v3 branch

https://docs.platformio.org/en/latest/projectconf/section_env_library.html

(using my phone, so hard to write out example)

MajorTwip, 2019-08-26:
https://github.com/MajorTwip/HomiePrimer/blob/master/platformio.ini as a sample. Just saw that i had defined multiple AJ...
Just look at the last line of the lib_deps.

MajorTwip, 2019-08-26:
Works fine for me with openHAB, actual Snapshot

psyciknz, 2019-08-27:
So the develop v3 version is pretty stable?  I converted to it last night just need to push out an new firmware.    Hopefully that means I can use it in openHAB now via discovery.

psyciknz, 2019-08-27:
I suspect my issue was I've moved development to a new machine and via platformio....and the homie library is 2.0.0 rather than 2.0.1....that might be the difference

psyciknz, 2019-08-28:
Yea I've gone all the way up to 3.0.0-develop - been running over night, will test it's still responding this afternoon.