homieiot/homie-esp8266 #635: Multiple Homie devices disconnect simultaneous

dakhnod, 2019-10-29:
I have a few Wemos D1 devices setup with some simple code, no mandatory library modifications and Homie.loop called regularly. 

If i am lucky, all devices stay online for a few weeks, but at one point all of them timeout basically within a few seconds.

I am opening this issue since i have not seen any other issue with a comparable setup and may be able to provide useful logs or similar.

Mosquitto does really not say anything helpful other than "exceeded timeout".

Is there any other interesting place i can extract logs of?

amayii0, 2019-10-29:
Hi

This is very strange, in the first place I'd say your router is the culprit.
Are you experiencing any other wifi issues with other devices? Smartphones maybe?
Looking again at network, maybe a routine restart/something of the router.
Which brand is it? Supplied by your ISP?

Do they all have the exact same up-time? I remember something about maximum value in some types (int and the likes) that could go bad.
For the test, it'd be nice to restart a few of your boards a few days later, to make sure they don't fail because of this. Effectively testing with different up-times.

CWempe, 2021-05-07:
I noticed a similar behavior with my Homie/Wemos devices.

But I am pretty sure this is caused by some kind of power issue.

It happens almost everytime I am using my 3D printer (Ender3) wich has a cheap power supply.
But occasionally happens when the printer is turned off, too.

After hitting the reset button of the devices they work fine again.

I have not further investigated this issue.

luebbe, 2021-05-09:
Disconnects were also caused by bugs in the async-mqtt library. A lot of them have been fixed in the last half year, so maybe it's worth to switch your devices to the development branch of async-mqtt.