homieiot/homie-esp8266 #637: Guru Meditation Error when connecting to wlan in config mode

dobernhardt, 2019-11-10:
When  trying to connect to the wlan with my notebook in config mode the board crashes with a Guru Mediation Error.

Tried on a ESP32 pico board as well as on an esp32-EVB board. With the pico board it worked once but on any subsequent connection attempt the board crashed.

Build develop-3 branch commit 71a766b with platformio.

PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x4011b6d2: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x401254c1: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
0x4012a4c9: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x4012f766: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x4011b89f: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

nemidiy, 2019-11-18:
hi @dobernhardt I think what you are experiencing is this :

https://github.com/homieiot/homie-esp8266/issues/588

There is a pull request done with a workaround otherwise what you can do is uploading the homie/config.json into the SPIFFS instead of booting into config mode.

You can use this tool to do so :
https://github.com/me-no-dev/arduino-esp32fs-plugin

dobernhardt, 2019-11-19:
Thanks for that hint. Yes that definitely looks like my scenario.

closed 2019-11-19 by dobernhardt