homieiot/homie-esp8266 #641: Commit 91e9b63 on develop-v3 causes crash on ESP32

nemidiy, 2019-11-24:
Hi guys, the last commit pushed to develop-v3 makes ESP32 crash. It does not crash on ESP2866 I tested it as well.

```log
commit 91e9b63c9618b3809a51ee7b0e0c057ad3bda62b (HEAD -> develop-v3, origin/develop-v3)
Author: peret2000 <19463568+peret2000@users.noreply.github.com>
Date:   Tue Nov 12 20:52:24 2019 +0100

    Subscribing to messages using Homie.getMqttClient() is not reliable: topic is truncated (#526)
    
    * Declare _mqttTopicCopy in BootNormal.hpp
    * Using _mqttTopicCopy to manage subscribed topics
    * Removed leading spaces
```

Backtrace :
```log
0x4008c800: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008ca31: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40088d1f: xQueueTakeMutexRecursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 621
0x400e5af8: spi_flash_op_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 56
0x400829ea: spi_flash_disable_interrupts_caches_and_other_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 95
0x40081b23: spi_flash_protected_read_mmu_entry at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_mmap.c line 299
0x400e5628: spi_flash_cache2phys at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_mmap.c line 426
0x400e8366: esp_ota_get_running_partition at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/app_update/esp_ota_ops.c line 508
0x400e5def: do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 444
0x40082cfb: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 358
0x40082f00: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 234
```

Just in case here is the sample code I am using :
```c++
#include <Homie.h>

HomieNode sensorNode("sensor", "Sensor", "sensor");

unsigned long lastSent = 0;

void loopHandler() {
  unsigned long m = millis();
  if ( m - lastSent >= 10 * 1000UL || lastSent == 0) {
    Homie.getLogger() << "Fake value: " << m << endl;
    sensorNode.setProperty("value").send(String(m));
    lastSent = millis();
  }
}

void setup() {

  Serial.begin(115200);
  Serial << endl << endl;

  
  Homie.setLoopFunction(loopHandler);
  Homie_setFirmware("test", "1.0.0");
  
  sensorNode.advertise("value").setName("Value").setDatatype("double");
  Homie.setup();
}

void loop(){
    Homie.loop();
}
```

And the platformio.ini 
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
upload_speed = 115200
lib_ldf_mode = deep
lib_deps =
    ArduinoJson@6.13.0
    https://github.com/homieiot/homie-esp8266.git#91e9b63c9618b3809a51ee7b0e0c057ad3bda62b

[env:nodemcu32]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
upload_speed = 115200
lib_ldf_mode = deep
lib_deps =
	ArduinoJson@6.13.0
   https://github.com/homieiot/homie-esp8266.git#91e9b63c9618b3809a51ee7b0e0c057ad3bda62b

;91e9b63c9618b3809a51ee7b0e0c057ad3bda62b -> crash on ESP32
;71a766bfef15fb066ff33fbce076e82d2f0a2d77 -> works
;9ce06e2b5fc5833d2d33c43120470452edc15908 -> works
```

luebbe, 2019-11-25:
Do you have any idea why it crashes? I merged the PR, because it did work on an ESP8266. I couldn't test on an ESP32

nemidiy, 2019-11-25:
@luebbe no, I don't. I guess this is the point in time where I learn how to use gdb stub on this things. I'll do my best and come back. It's going to take some time though, lol.

luebbe, 2019-11-26:
Related to #526

nemidiy, 2019-11-27:
Found the problem. Here is the PR : https://github.com/homieiot/homie-esp8266/pull/642

sradner13, 2019-11-29:
Can anyone verify this also addresses the OTA update watchdog exceptions?  I'm away at the moment, but could run some tests on Dec, 2nd.

closed 2019-12-02 by nemidiy

sradner13, 2019-12-04:
Tested with latest library.  This patch does not address changes to OTA updates.  esp8266 consistently crashes on OTA with a watchdog timer error.  If I revert to the code prior to the introduction of _mqttTopicCopy  I am once again able to update OTA.

nemidiy, 2019-12-04:
hi @sradner13, I did not know that OTA was borken as well. Actually, when I sent the patch my focus was getting ESP32 to boot (since ESP8266 had no issues booting). Yesterday I created #644  since I started testing out OTA and it fails both on ESP8266 as well as ESP32. So I am going to check what you are saying and maybe we can keep track of things in #644 since it's OTA specific?