homieiot/homie-esp8266 #648: OTA with MQTT TLS not working

nemidiy, 2020-01-02:
Hi everyone, happy new year!

I finally got to the point where I could test OTA updates under TLS. As we know it does not work.
Doc gives a hint :

> ASYNC_TCP_SSL_ENABLED
> This compiler flag allows to use SSL encryption for MQTT connections. All other network 
> connections still can not be encrypted like HTTP or OTA.

What was rather painful to find is that for esp32 [AsyncTcp](https://github.com/me-no-dev/AsyncTCP) has no support for TLS. There is a PR though that includes client side TLS using an mbed lib. Since all this adds an extra level of complexity I-ll start by debugging on ESP8266 and then if I can make that work Ill see what I can do in esp32 (my board of choice).

If anyone has any ideas on what the problem is for OTA not working on TLS please let me know :)

Here is what I am doing :
```shell
python ota_updater.py -l mqtt.dc-iot.com -p 8883 -t devices/ -i testbox8266 --broker-tls-cacert ca.crt /home/nemi/workspace/dc/homie_test/firmware.bin
Connecting to mqtt broker mqtt.dc-iot.com on port 8883
Connected with result code 0
Waiting for device to come online...
Waiting for device info...
Publishing new firmware with checksum 367e1be9f437c66f747a54de50016720
Waiting for device info...
Expecting checksum 367e1be9f437c66f747a54de50016720, got fc9d1211fbc105e51d3e4359d76cd108, update failed!
```
mqtt.dc-iot.com resolves in my private DNS to my mqtt server, that-s not even a real domain.
I can see if I subscribe to the base topic (using mosquitto_sub) that the FW is pushed just fine but then when the board receives the message it just disconnects from the server

```shell
💡 Firmware test (1.0.0)
🔌 Booting into normal mode 🔌
SSL is: 1
Using fingerprint: 2362f4fbc7890e450d43bb377220e2ac85409936
{} Stored configuration
  • Hardware device ID: cc50e33ca8c5
  • Device ID: testbox8266
  • Name: Test Box 8266
  • Device Stats Interval: 10 sec
  • Wi-Fi: 
    ◦ SSID: dc-iot
    ◦ Password not shown
    ◦ IP: 192.168.0.201
    ◦ Mask: 255.255.255.0
    ◦ Gateway: 192.168.0.1
  • MQTT: 
    ◦ Host: mqtt.dc-iot.com
    ◦ Port: 8883
    ◦ SSL enabled: true
    ◦ Fingerprint: 2362f4fbc7890e450d43bb377220e2ac85409936
    ◦ Base topic: devices/
    ◦ Auth? no
  • OTA: 
    ◦ Enabled? yes
...
〽 Sending statistics...
  • Interval: 15s (10s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 597s
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
〽 Sending statistics...
  • Interval: 15s (10s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 599s
```

here is my platformio ini file
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags =
    ;-D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -D ASYNC_TCP_SSL_ENABLED=1
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
upload_speed = 115200
lib_ldf_mode = deep
lib_deps =
    ArduinoJson@6.13.0
    https://github.com/homieiot/homie-esp8266.git#develop-v3
```

and the deps :
```shell
|-- <ArduinoJson> 6.13.0
|-- <Homie> 3.0.0 #104a09c
|   |-- <ArduinoJson> 6.13.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <Bounce2> 2.52
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.13.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Ticker> 1.0
```

nemidiy, 2020-01-04:
The problem seems to be in ESPAsyncTCP. I created an issue on that project to see if I get some pointers. Here it is : https://github.com/me-no-dev/ESPAsyncTCP/issues/133

nemidiy, 2020-01-10:
Ok, got no answers at the ESPAsyncTCP. Good news is that I have Homie working for ESP32 with TLS and starts doing OTA over TLS (I say "starts" because it crashes eventually). It's a work in progress but here is what I have accomplished so far ...

Using a PR branch on AsyncTCP to merge in mbed-tls and propagating some changes on AsyncMqttClient and Homie I have my ESP32 connecting to mosquitto using TLS with pre shared keys. In fact you can try it out yourselves :

```ini
[env:nodemcu32]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
build_flags =
    -D ASYNC_TCP_SSL_ENABLED=1
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH  
upload_speed = 115200
lib_ldf_mode = deep
lib_deps =
   ArduinoJson@6.13.0
   https://github.com/nemidiy/AsyncTCP.git#mbed-tls-try2
   https://github.com/nemidiy/async-mqtt-client.git#mbed-tls
   https://github.com/nemidiy/homie-esp8266.git#feature/mbed-tls-esp32
```

The PSK and PSK-IDENT need to be added to your config :
```json
{
  "name": "Test Box 32",
  "device_id": "testbox32",
  "device_stats_interval": 10,
  "wifi": {
    "ssid": "YYYYYYYYYYYYY",
    "password": "XXXXXXXXXX",
    "ip":"192.168.0.202",
    "gw":"192.168.0.1",
    "mask":"255.255.255.0",
    "dns1":"192.168.0.232"
  },
  "mqtt": {
    "host": "mqtt.dc-iot.com",
    "port": 9883,
    "base_topic": "devices/",
    "ssl":true,
    "psk_identity":"test", -> new setting
    "psk":"XXX" -> new setting
  },
  "ota": {
    "enabled": true
  }
}
```

This is how the mosquitto listener has been set up:
```ini
listener 9883
allow_anonymous true
psk_hint somecrazyhint
psk_file /etc/mosquitto/conf.d/psk.txt
```

Here is the output for the serial :
```ini
💡 Firmware test (1.0.0)
🔌 Booting into normal mode 🔌
SSL is: 1
{} Stored configuration
  • Hardware device ID: 30aea41c0018
  • Device ID: testbox32
  • Name: Test Box 32
  • Device Stats Interval: 10 sec
  • Wi-Fi: 
    ◦ SSID: dc-iot
    ◦ Password not shown
[... 110 lines cut ...]
Receiving OTA firmware (831403/1281392)...
Receiving OTA firmware (832427/1281392)...
Receiving OTA firmware (833451/1281392)...
Receiving OTA firmware (834475/1281392)...
Receiving OTA firmware (835499/1281392)...
se r� tn  • Wi-Fi signi.o nn  n "�n de dw_ n rncv n _n a� nnn n_ n wn n�d  <d=    d0 dxf n fn ff�" n    fn an in ln en dn :n  n f�il de  n" n /n Un sn e�r sn /f�i dc en tn on /nD ne n sn kn tn o� pn/ dES Pd 3n2 /dE dS Pn 3n2 /ne dsp �- in dS fn- pdu dib ln in cn/ dco n mn pn on nne nn t�s /�l nw ni dp /n ln wi�p /e sn r�c /d c nor� en /n tn c  pn. nc n" d, l ni dn � en  d77 9d ,n  dfu dn cn tn in oen d:  nt nc np �_ un p�d ant de d _n r�c vn _a�n nd_ dw d nn dn
 Sbort( ) was calsled at PC 0x400e3063non co1e 0

Backtrace: 0x4008c800:0x3ffb5370 0x4008ca31:0x3ffb5390 0x400e3063:0x3ffb53b0 0x400feebe:0x3ffb53e0 0x400fef4d:0x3ffb5400 0x4018bb75:0x3ffb5420 0x400fcbe0:0x3ffb5440 0x4008b015:0x3ffb5470

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
```

Have to figure out why it crashes ... but given that we had no TLS support on ESP32 and that OTA was not even beginning to work with TLS I am quite happy so far. I'll keep working on it an get back.

kleini, 2020-01-13:
Will test as soon as possible on ESP8266 with SSL encrypted MQTT connection.

nemidiy, 2020-01-13:
hey @kleini thanks for jumping in. Nothing I have done so far applies to ESP8266, just ESP32

nemidiy, 2020-01-13:
[UPDATE] Found why OTA crashes, not quite get the underlying reason to be honest, regardless I
was able to fix the code to have OTA under TLS stable enough.

Contex when doing OTA :
  * we get all the firmware in an encrypted channel
  * for each processed chunk of the received firmware a status message is sent
  * we keep sending stats and also whatever the user wants to publish
  
That puts to much pressure on the AsyncTCP lib and it gets  to a state where the  TCP  window 
on the device goes down to zero. I was able to confirm this because during all this tasks the publish
method for the async-mqtt-client libe exits here :

https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L862

and after doing so, it just crashes. I took the following approach and got it working...

while the OTA update is ongoing:
* stop sending stat messages
* do not invoke the user defined loop function to avoid doing any user level messaging.
* publish 1 out 100 of the OTA 206 messages.

Still if I do a trace I can see the ESP32 window going down to zero but at least works.
https://imgur.com/a/FyPxGD3

I pushed this fixes to the branches in the platformio.ini so if anyone has an ESP32 laying around I 
would appreciate someone tries the OTA over TLS and lets me know :) I would like to rule out any 
environmental issues in my home network.

Here is the main I am using :
```c++
#include "Arduino.h"
#include <Homie.h>

HomieNode sensorNode("sensor", "Sensor", "sensor");

unsigned long lastSent = 0;

void loopHandler() {
  unsigned long m = millis();
  if ( m - lastSent >= 10 * 1000UL || lastSent == 0) {
    Homie.getLogger() << "Fake value: " << m << endl;
    sensorNode.setProperty("value").send(String(m));
    lastSent = millis();
  }
}

void setup() {
  Serial.begin(115200);
  Homie_setFirmware("test", "1.0.0");
  Homie.setLoopFunction(loopHandler);
  sensorNode.advertise("value").setName("Value").setDatatype("double");
  Homie.setup();
}

void loop(){
  Homie.loop();
}
```

kleini, 2020-01-14:
Great work @nemidiy !!! This is the first time, OTA worked for me over a SSL connection.
Your code with ESP8266 fixes: https://github.com/kleini/homie-esp8266/commits/ota
Serial log from device:
```
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 86%
  • Uptime: 246s
Receiving OTA payload
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary
Receiving OTA firmware (16295/437808)...
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Receiving OTA payload
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary
Receiving OTA firmware (16295/437808)...
Receiving OTA firmware (32679/437808)...
Receiving OTA firmware (49063/437808)...
Receiving OTA firmware (65447/437808)...
Receiving OTA firmware (81831/437808)...
Receiving OTA firmware (98215/437808)...
Receiving OTA firmware (114599/437808)...
Receiving OTA firmware (130983/437808)...
Receiving OTA firmware (147367/437808)...
Receiving OTA firmware (163751/437808)...
Receiving OTA firmware (180135/437808)...
Receiving OTA firmware (196519/437808)...
Receiving OTA firmware (212903/437808)...
Receiving OTA firmware (229287/437808)...
Receiving OTA firmware (245671/437808)...
Receiving OTA firmware (262055/437808)...
Receiving OTA firmware (278439/437808)...
Receiving OTA firmware (294823/437808)...
Receiving OTA firmware (311207/437808)...
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Receiving OTA payload
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary
Receiving OTA firmware (16295/437808)...
Receiving OTA firmware (32679/437808)...
Receiving OTA firmware (49063/437808)...
Receiving OTA firmware (65447/437808)...
Receiving OTA firmware (81831/437808)...
Receiving OTA firmware (98215/437808)...
Receiving OTA firmware (114599/437808)...
Receiving OTA firmware (130983/437808)...
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Receiving OTA payload
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary
[... 460 lines cut ...]
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 9s
```
and ota_updater logs
```
$ ota_updater.py
Connecting to mqtt broker 192.168.0.6 on port 1883
Connected with result code 0
Waiting for device to come online...
Waiting for device info...
Publishing new firmware with checksum cd968a8b37a7d5c0c8ca338533e7e347
Expecting checksum cd968a8b37a7d5c0c8ca338533e7e347, got 6c80883d4703a310907d8bfdf985e9c1, update failed!
$ ota_updater
Connecting to mqtt broker 192.168.0.6 on port 1883
Connected with result code 0
Waiting for device to come online...
Waiting for device info...
Publishing new firmware with checksum cd968a8b37a7d5c0c8ca338533e7e347
[++++++++++++++++++++++++++    ] 393127/437808Device back online. Update Successful!
```

nemidiy, 2020-01-14:
@kleini that's great news! so if what I did on homie + async-mqtt improved ESP8266 as a side effect (and it was pure luck since I was targetting ESP32) it probably means that the ESPAsynTCP lib that uses a axtls ssl lib is also having issues with buffers that probably generate those MQTT disconnections due to timeouts. I am currently trying to build the binary using esp-idf since the sdkconfig.h allows to specify the underlying TCP window size. Thanks for the feedback. Hopefully I'll have more news soon.

kleini, 2020-01-15:
Oh, I missed your changes on AsyncMqttClient. It worked without that. After having a look into your changes, those are only for ESP32, I don't see any improvements for ESP8266. Furthermore I don't see, where axtls is replaced with mbedtls. But even without that, it is a big step forward for me.

nemidiy, 2020-01-16:
@kleini yes. you are right about AsyncMqttClient. axtls is only replaced by mbedtls only if use espressif32 as the platform since dependencies in AsyncMqttClient are :
```
"dependencies": [
    {
      "name": "ESPAsyncTCP",
      "version": "1.2.0",
      "platforms": "espressif8266"
    },
    {
      "name": "AsyncTCP",
      "version": "^1.0.0",
      "platforms": "espressif32"
    }
  ]
```
the branch for AsyncTCP in the ini file is the one that replaces axtls by mbed. Anyways I am still struggling with platformio using the arduino core as an isp-idf component in order to play with window
sizes and other mbed configs, fun and painful at the same time.

kleini, 2020-01-16:
Fiddling around with TLS implementations is currently too much for me. My C++ skills are too unexercised for that. But at least your changes make OTA possible for my devices. If I can further support you in getting more things tested and done, just tell me, how I can help.

nemidiy, 2020-01-26:
Ok, so I built homie and all it's dependencies using esp-idf for the ESP32 platform to see if playing around with the TCP window size, would remove the window dropping down to 0 from time to time when doing the OTA updates over TLS. Short story, did not work ...
* tried different TCP windows sizes
* tried different TLS message max sizes
* tried having AsyncTCP using 1 core and both cores
* tried some mbed specific configs 

I just concluded that the window dropping down to zero is just the producer (broker) being much faster than the consumer (MCU) when doing too much decryption as needed for the OTA

Anyways, and regardless the window dropping to zero:
*  OTA over TLS using mbedTLS on ESP32 is stable which is the good news.
*  OTA over TLS under axtls on ESP8266 is working now with the tweaks in the homie branch and that @kleini tested out.

For ESP8266 since the changes are just in the homie repo I can separate them from the ESP32 and create a PR.

For ESP32 I first need :
1. https://github.com/me-no-dev/AsyncTCP/pull/48 getting merged merged, then ...
2. then do a PR in the async-mqtt-client project to add the PSK support, then ...
3. Homie PR

Thus ... stay tunned for "TLS tweaks" PR during the week. Thanks

robert-alfaro, 2020-01-26:
Do you have any issues with closing the AsyncTCP connection or do you simply auto-reboot upon completion (or failure) and avoid the bug? I am not using mqtt, but am using AsyncTCP with mbedTLS on ESP32 pulling a payload from GCP. All is great, no issues, except closing the connection.

nemidiy, 2020-01-26:
hey @robert-alfaro, correct after OTA updating the device just reboots into the new firmware. Unless there is a problem on MQTT, the socket does not get closed, and so far I have not had any MQTT disconnections, but I can try it out and see what happens. Could you tell me about your use case so I can try and reproduce it ? Meaning, is the socket getting closed by your code because you are done doing whatever you need or is it closing due to a problem in the channel (ie, latency or whatever) ?

BTW, all my tests are in an internal network, and if I do any connections with the outside world I do so through an MQTT bridge (mosquitto) since I am paranoid about exposing devices and I can run the bridge in a more powerful and secured device (ie raspberry).

Also .. are you not getting this ? ... cant figure this log out.
```
[E][AsyncTCP.cpp:1130] _poll(): 0x3ffcc148 != 0x3ffd293c
```

luebbe, 2020-02-12:
Wow, 4/4 approvals for the PR. I guess, we can close this issue.

closed 2020-02-12 by luebbe