homieiot/homie-esp8266 #651: WiFi loss

nelfata, 2020-01-06:
What is the expected behavior if the WiFi connection was lost or the MQTT server no longer responds?
In my initial test, it seems like upon WiFi connection loss the device no longer performs any task (I had a simple automation setup to turn on an LED upon a button press). So the button press was no longer responsive.
I think that should never occur, the WiFi loss should not interfere with the programmed automation/task. In addition it should be able to recover and should not require a power cycle.

mwildbolz, 2020-01-06:
Seems to me like this is already discussed in #490

euphi, 2020-01-06:
By default the loop function is only called if connected to MQTT.

But you can use   `setRunLoopDisconnected(true)` on/in your HomieNode class to override this default behaviour and also run your loop in disconnected state.

closed 2020-01-06 by euphi

nelfata, 2020-01-07:
Ok I will give it a try, I moved on to use ESPHOME. Thank you.