homieiot/homie-esp8266 #655: callback function handleInput not working for me

ElizabethMorves, 2020-01-10:
Hi guys. can't turn on callback.
esp8266(nodemcuv2)/esp32devkit v4, homie 3.0.1, external mqtt broker mosquitto, wifi, no ssl.
A slightly modified ping example. the original also does not work.
so simple/little code.
```c++
#include "LoggerNode.h"
#include "pingNode.h"

HomieSetting<long> pingNode::pingInterval("pingInterval", "Ping execute interval");
pingNode::pingNode(const char* id, const char* name) : HomieNode(id, name, "ping"),
lastPingLoop(0){
    pingInterval.setDefaultValue(30000).setValidator([] (long candidatePing) {
    return (candidatePing > 1000 && candidatePing < 600000);
    });
}

bool pingNode::handleInput(const HomieRange& range, const String& property, const String& value){
  LN.logf(__PRETTY_FUNCTION__, LoggerNode::DEBUG, "Received  property %s (value=%s).", property.c_str(), value.c_str());
  Homie.getLogger() << "  ◦ " << "〽 handleInput -> property '" << property << "' value=" << value << endl;
  setProperty("ping").send("pong");
  return true;
}

void pingNode::loop() {
  if (millis() - lastPingLoop >= (unsigned long) pingInterval.get() || lastPingLoop == 0) {
    lastPingLoop = millis();

    Homie.getLogger() << "〽 Sending ping 'hello': " << getId() << endl;
    setProperty("ping").send("hello");
  }
}

void pingNode::setup() {
    advertise("ping").setName("Ping").setRetained(true).setDatatype("string").settable();
}

```
Tell me how to make it work? Thank you for your time.

euphi, 2020-01-10:
Most common cause for issues like these is sending on the wrong topic.

According to the convention you must send the value to

homie/_<device_id>_/ping/ping/**set**

Often people forget the `set`.... Without set, homie-esp8266 does not call the input handler.

ElizabethMorves, 2020-01-10:
Many thanks. This is good news for me. Now I can move on.
Its so good =)  **set**
```
564716 [DEBUG]: virtual bool pingNode::handleInput(const HomieRange&, const String&, const String&): Received  property ping (value=hello).
  ◦ 〽 handleInput -> property 'ping' value=hello
〽 Sending ping 'hello': ping
574591 [DEBUG]: virtual bool pingNode::handleInput(const HomieRange&, const String&, const String&): Received  property ping (value=hello).
  ◦ 〽 handleInput -> property 'ping' value=hello
```

closed 2020-01-10 by ElizabethMorves