homieiot/homie-esp8266 #661: Config file is not valid, reason: mqtt.password is too long

spacemanspiff2007, 2020-02-13:
Hello,

my mqtt broker password is 40 digits long. I the password to the config file and uploaded it to the device as described [here](https://homieiot.github.io/homie-esp8266/docs/3.0.0/configuration/json-configuration-file/).
However, I receive the following error:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
✖ Config file is not valid, reason: mqtt.password is too long
Configuration invalid. Using CONFIG MODE
```

What is the maximum allowed mqtt password length?

Could this be added to the documentation page linked above?

Thank you!

ech89899, 2020-02-13:
@spacemanspiff2007 I also faced the same issue.
Homie's limits for MQTT broker username and password are `32+1` symbols.
You can find it `MAX_MQTT_CREDS_LENGHT` constant in the `Limit.hpp` file.
I just updated it with `64+1` value, that resolved issue in my case.

spacemanspiff2007, 2020-02-13:
Do you know how to add this to the documentation?
I think it'll be really useful for the future.

closed 2020-02-19 by spacemanspiff2007