homieiot/homie-esp8266 #663: Homie.setLedPin does not seem to work

spacemanspiff2007, 2020-02-13:
Hi,

I have a NodeMCU with an ESP32 which has the built in LED on GPIO2 however I can not make it work.
I have configured a wrong mqtt user and according to the [getting started](https://homieiot.github.io/homie-esp8266/docs/3.0.0/quickstart/getting-started/#visual-codes) I'd expect it to indicate that it's connecting to the broker.
However after the initial light up (just to see if I have the right gpio) nothing happens and the LED stays dark. Am I doing something wrong?
Thanks for the help!

````c++
#include <Homie.h>

#define LED_BUILTIN 2

HomieSetting<long> LEDcount("leds", "How many LEDs are connected");  // id, description

void setup() {
  Serial.begin(115200);

  // NodeMCU hat 2te LED an GPIO2
  pinMode(LED_BUILTIN, OUTPUT);
  Homie.setLedPin(LED_BUILTIN, HIGH);

  // this will light the led up
  digitalWrite(LED_BUILTIN, HIGH);
  delay(300);
  digitalWrite(LED_BUILTIN, LOW);

  // validate LED count
  LEDcount.setDefaultValue(1).setValidator([] (long candidate) {
    return (candidate > 0);
  });

  Homie_setFirmware("Test", "0.0.1"); // The underscore is not a typo! See Magic bytes

  Homie.setup();
}

void loop() {
  Homie.loop();
}
````

````
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
💡 Firmware Test (0.0.1)
🔌 Booting into normal mode 🔌
{} Stored configuration
...
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.199
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
````

codefrey, 2020-03-06:
LED is disabled by default on the ESP32 platform (see https://github.com/homieiot/homie-esp8266/blob/a22f7e3ac483993169c511ee063e165a28d3fac1/src/Homie.cpp#L13) and there does not appear to be a way to enable it.

I will try to implement a feature like this later.

spacemanspiff2007, 2020-03-10:
Thank you for your efforts!

PR #667 (2020-03-30): Enable LED feedback when LED pin is set
Implicitly enable LED feedback when a LED pin is set to allow using the LED pin on the ESP32 platform.

Fixes #663
closed 2020-03-30 by luebbe