homieiot/homie-esp8266 #666: Reset configuration not working

Ubus99, 2020-03-06:
i need to change the network configuration of my esp8266, but neither holding reset for 5S, nor
`#include <Homie.h>

void setup() {
  // put your setup code here, to run once:
  Homie.reset();
  Homie.setup();
}

void loop() {
  // put your main code here, to run repeatedly:

}`
work.
the esp8266 will enter a state in which it will remain in "normal" mode forever, but with a greatly reduced log.

codefrey, 2020-03-06:
Can you paste the log output?

labodj, 2020-04-04:
Try

```c++
#include <Homie.h>
void setup() {
  Homie.setup();
  Homie.reset();
  Homie.setIdle(true);
}

void loop() {
}
```

coal2001, 2020-04-09:
Also doesnt' work resetting through mqtt on topic $implementation/reset

https://homieiot.github.io/homie-esp8266/docs/stable/others/homie-implementation-specifics/

I don't know if that is the same bug