homieiot/homie-esp8266 #668: Can't read I2C sensor when MQTT is connected.

cafe01, 2020-03-15:
Hello,
I have implemented a simple temperature/humidity device using NodeMCU and BME280 I2C sensor. 
The hardware is working properly as I could confirm running a simple firmware without Homie.
After porting to Homie, I could not read the sensor anymore. It would only return nan.
I've tried to call the loopHandler() function by myself inside loop(). Didn't work either.
For some reason, I've restarted the mosquitto server and saw the correct sensor readings on the serial monitor.

After that I've stopped and started the mqtt server many times. The I2C reading worked flawlessly when mqtt was disconnected, and stopped working every time after the MQTT_READY event.

Please, help! :)

Serial Monitor:
```💡 Firmware test-homie (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: 807d3a3c8d06
  • Device ID: 807d3a3c8d06
  • Name: IPCC
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: VIVO-26E0
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.0.7
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA: 
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
nan | nan
26.98 | 73.89
26.96 | 73.66
26.96 | 73.57
✔ Wi-Fi connected, IP: 192.168.0.12
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
26.99 | 74.12
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 5s
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 6s
nan | nan
nan | nan
nan | nan
nan | nan
✖ MQTT disconnected, reason: 0
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
27.00 | 73.84
↕ Attempting to connect to MQTT...
27.00 | 73.26
27.00 | 73.15
27.00 | 72.92
27.00 | 72.88
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 23s
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 100%
  • Uptime: 23s
nan | nan
nan | nan
nan | nan
```

platformio.ini
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
lib_deps = BME280 , https://github.com/homieiot/homie-esp8266.git#develop

monitor_speed = 9600
upload_speed = 921600
```
main.src
```
#include <Homie.h>
#include <Wire.h>
#include <SPI.h>
#include <BME280I2C.h>

BME280I2C::Settings settings(
    BME280::OSR_X1,
    BME280::OSR_X1,
    BME280::OSR_X1,
    BME280::Mode_Forced,
    BME280::StandbyTime_1000ms,
    BME280::Filter_Off,
    BME280::SpiEnable_False,
    BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);

BME280I2C bme(settings);

unsigned long last_loop = 0;

void loopHandler() {

  if (millis() - last_loop > 1000) {
    last_loop = millis();
    Serial.printf("%.2f | %.2f\n", bme.temp(), bme.hum() );
  }

}

void setup() {

  Serial.begin(9600);
    
  Wire.begin(D3, D4);

  if (!bme.begin()) {
      Serial.println("Could not find a valid BME280 sensor, check wiring!");
  }

  Homie_setFirmware("test-homie", "1.0.0");
  // Homie.setLoopFunction(loopHandler);
  Homie.setup();
}

void loop() {
  Homie.loop(); 
  loopHandler();
}
```

kleini, 2020-03-16:
I don't think, it is a good idea to read the sensor values in every `loop()` method run. Reading I²C values takes some time and the fast execution of the `loop()` method is essential to keep TCP connections alive. The ESP device only has the possibility to answer to TCP packets within this `loop()` method and nowhere else. If the correct time windows for answering TCP packets are missed because the above code is reading I²C values, this causes those connections problems, you're seeing. Once the TCP connections are correctly established and don't transfer packets all the time any more, you can start reading I²C values in every `loop()` run. But this does not make sense either, it just burns CPU cycles, consumes power and is a potential problem for keeping all the other tasks of Homie alive.

closed 2020-03-16 by kleini

cafe01, 2020-03-16:
I'm not having connection problems. 

I'm reading the I2C sensor at 1 second interval from the loopHandler (called from Homie). 
(Just comment the `loopHandler();` in `loop()` and uncomment `Homie.setLoopFunction(loopHandler);` in `setup()`.

The reading doesn't work when MQTT is connected. 
The reading works when MQTT disconnected.

Again, I'm not having mqtt connection problem. I'm manually stopping/startting the mosquitto server.

kleini, 2020-03-16:
Then the problem is maybe more in the BME library implementation than in Homie.

RobAxt, 2020-03-16:
´[env:d1_mini]
platform = espressif8266
board = d1_mini´

RobAxt, 2020-03-16:
Hello cafe01,
I'm working with a Wemos D1 mini, not a NodeMCU like you. But I notice you are using D3 and D4 and those pins are used by Homie for reset and Led.
Hope this help!!

luebbe, 2020-03-17:
@cafe01 please check, if the following example works for you.
https://github.com/luebbe/homie-node-collection/blob/master/examples/demo-sensor-nodes.cpp
You can comment out the sensors that you don't have, but they don't cause any harm if you leave them in.

cafe01, 2020-03-17:
Thank you @kleini, @RobAxt, @luebbe for your attention! Looks like D4 was the culprit. A beginner mistake. 
I'ts working now, I've set Homie to use the other onboard led on GPIO16. (`Homie.setLedPin(16, LOW);`)
Thanks @luebbe for that repo, very useful to learn code organization.