homieiot/homie-esp8266 #670: Can't get homie-esp8266 to work with Neopixels

tavalin, 2020-03-22:
Hi,

I'm using version 3.0.0 of homie-esp8266 and I've created the following sketch where my intention is to have a homie device with an led node where I can set the color property.  This color property should change a Neopixel accordingly:

```
#include <Homie.h>
#include <Adafruit_NeoPixel.h>

#define FW_NAME "test-fw"
#define FW_VERSION "1.0.0"

#define LED_COUNT 1
#define LED_PIN D3
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

String color = "off";
long last = 0;

HomieNode ledNode("led", "led", "led");

bool ledHandler(HomieRange range, String value) {
  Serial << value << endl;
  color = value;
  color.trim();
  return true;
}

void setup() {
  Serial.begin(115200);
  Serial.println(FW_NAME FW_VERSION);

  Homie_setFirmware(FW_NAME, FW_VERSION);

  ledNode.advertise("color").setDatatype("color").settable(ledHandler);
  
  strip.begin();
  strip.setPixelColor(0, strip.Color(0, 150, 0));  // This sets the neopixel to green
  strip.show();

  Homie.setup();
}

void loop() {
  Homie.loop();
  processColor();
}

void processColor() {
  if (millis() - last > 1000) {
    last = millis();
    Serial << "[processColor()] " << " Start" << endl;
    Serial << "[processColor()] " << " color = " << color << endl;

// This section currently does not change the Neopixel color
    if (color == "off") {
      strip.setPixelColor(0, strip.Color(0, 0, 0));
    } else if (color == "red") {
      strip.setPixelColor(0, strip.Color(255, 0, 0));
    } else if (color == "green") {
      strip.setPixelColor(0, strip.Color(0, 255, 0));
    } else if (color == "blue") {
      strip.setPixelColor(0, strip.Color(0, 0, 255));
    }
    strip.show();
    delay(500);
    Serial << "[processColor()] " << strip.getPixelColor(0) << endl;
  }
}
```

For some reason, once Homie connects to the MQTT broker, the Neopixels no longer respond to commands.

Has anyone got any ideas why this is the case?

Thanks

kleini, 2020-03-23:
Calling `delay` should be good on one hand, as the ESP8266 Arduino code is then able to do its background tasks for WiFi and TCP - at least I read about that, that Arduino code is using it this way. On the other hand, delaying the general `loop` method for 500ms is a very bad idea. As shown in the above example, the `loop` method can only run twice within a second. This will definitely break Homie code and its used libraries. I am sure, they require faster cycles in the `loop` method.

tavalin, 2020-03-23:
Thanks for your suggestion. I've tried the delay both with values like 10ms, as well as 500ms and neither work. 

If I comment out the lines for `Homie.setup()` and `Homie.loop()` the LED driver code works again so it seems the library is doing something fundamental that is affecting the ability to drive the LEDs - maybe it is affecting the timing on the LED signal PIN?

kleini, 2020-03-23:
https://github.com/adafruit/Adafruit_NeoPixel mentions a `clear` method. You didn't use it. Maybe that is the cause but I have no experience with the libraries nor that I read the source code.
But you should remove the call to the `delay` method in the `loop`. This is essential for Homie.

tavalin, 2020-03-23:
Just to be clear, even with that delay, Homie is still connected to the MQTT broker and still reporting statistics.

The problem is, the inclusion of Homie affects the ability of both the NeoPixel and FastLED libraries to drive the LEDs.

Here is another example which doesn't work once I've called `Homie.setup()` and `Homie.loop()`.

```
// A basic everyday NeoPixel strip test program.

// NEOPIXEL BEST PRACTICES for most reliable operation:
// - Add 1000 uF CAPACITOR between NeoPixel strip's + and - connections.
// - MINIMIZE WIRING LENGTH between microcontroller board and first pixel.
// - NeoPixel strip's DATA-IN should pass through a 300-500 OHM RESISTOR.
// - AVOID connecting NeoPixels on a LIVE CIRCUIT. If you must, ALWAYS
//   connect GROUND (-) first, then +, then data.
// - When using a 3.3V microcontroller with a 5V-powered NeoPixel strip,
//   a LOGIC-LEVEL CONVERTER on the data line is STRONGLY RECOMMENDED.
// (Skipping these may work OK on your workbench but can fail in the field)

#include <Adafruit_NeoPixel.h>

#include <Homie.h>
#define FW_NAME "test-fw"
#define FW_VERSION "1.0.0"

#ifdef __AVR__
 #include <avr/power.h> // Required for 16 MHz Adafruit Trinket
#endif

// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN    D3

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT 1

// Declare our NeoPixel strip object:
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
// Argument 1 = Number of pixels in NeoPixel strip
// Argument 2 = Arduino pin number (most are valid)
// Argument 3 = Pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

// setup() function -- runs once at startup --------------------------------

void setup() {
  // These lines are specifically to support the Adafruit Trinket 5V 16 MHz.
  // Any other board, you can remove this part (but no harm leaving it):
#if defined(__AVR_ATtiny85__) && (F_CPU == 16000000)
  clock_prescale_set(clock_div_1);
#endif
  // END of Trinket-specific code.

  strip.begin();           // INITIALIZE NeoPixel strip object (REQUIRED)
  strip.show();            // Turn OFF all pixels ASAP
  strip.setBrightness(50); // Set BRIGHTNESS to about 1/5 (max = 255)

  // This works and sets the Neopixel accordingly
  colorWipe(strip.Color(255,   0,   0), 0); // Red

  // Add Homie Setup
    Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie.setup();
}

// loop() function -- runs repeatedly as long as board is on ---------------
[... 75 lines cut ...]
  }
}

// Rainbow-enhanced theater marquee. Pass delay time (in ms) between frames.
void theaterChaseRainbow(int wait) {
  int firstPixelHue = 0;     // First pixel starts at red (hue 0)
  for(int a=0; a<30; a++) {  // Repeat 30 times...
    for(int b=0; b<3; b++) { //  'b' counts from 0 to 2...
      strip.clear();         //   Set all pixels in RAM to 0 (off)
      // 'c' counts up from 'b' to end of strip in increments of 3...
      for(int c=b; c<strip.numPixels(); c += 3) {
        // hue of pixel 'c' is offset by an amount to make one full
        // revolution of the color wheel (range 65536) along the length
        // of the strip (strip.numPixels() steps):
        int      hue   = firstPixelHue + c * 65536L / strip.numPixels();
        uint32_t color = strip.gamma32(strip.ColorHSV(hue)); // hue -> RGB
        strip.setPixelColor(c, color); // Set pixel 'c' to value 'color'
      }
      strip.show();                // Update strip with new contents
      delay(wait);                 // Pause for a moment
      firstPixelHue += 65536 / 90; // One cycle of color wheel over 90 frames
    }
  }
}
```

kleini, 2020-03-23:
Do you use the 3.3V to 5V logic level converter? Maybe 3.3V are enough for doing something initial on the LED strip, but once, WiFi and MQTT are connected and more power is consumed, the 3.3V level is not enough any more.

tavalin, 2020-03-23:
Thanks again for your latest suggestion - I wasn't using a level converter so I decided to give that a try with a 74AHCT125 I had spare.  

Unfortunately even with the level converter the results are the same - my code will only control the LEDs before `Homie.setup()` and `Homie.loop()` are called.

tavalin, 2020-03-23:
I've put together a new sketch which uses the PubSub MQTT client instead, which can successfully change the LED colour when a message is received.  This shows that this is not a fundamental issue due to the basic ESP WiFi connection stealing CPU cycles away from the LED driver - to me this very much points to something under the hood in the `Homie.setup()` and `Homie.loop()` code causing the issue, though I'm not familiar enough with the code to understand what exactly is causing this.

Here's the sketch that connects to the broker and receives messages:
```
#include <Adafruit_NeoPixel.h>
#include <PubSubClient.h>
#include <ESP8266WiFi.h>

WiFiClient espClient;
PubSubClient client(espClient);

const char* ssid = "Sailor_Jerry";
const char* password = "Donkeybreath1958";
const char* mqtt_server = "192.168.0.44";

////////////////////////////////
// LED
////////////////////////////////

#define LED_COUNT 64
#define LED_PIN D3
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

uint8_t ledBrightness = 255;

long lastReconnectAttempt = 0;

void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  payload[length] = '\0';
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.print(" - ");
  Serial.print(length);
  Serial.println();

  String color = String((char*)payload);

  if (color == "off") {
    strip.fill(strip.Color(0, 0, 0));
  } else if (strcmp((char*)payload, "red") == 0) {
    Serial.println("in red");
    strip.fill(strip.Color(255, 0, 0));
  } else if (strcmp((char*)payload, "green") == 0) {
    Serial.println("in green");
    strip.fill(strip.Color(0, 255, 0));
  } else if (strcmp((char*)payload, "blue") == 0) {
    Serial.println("in blue");
    strip.fill(strip.Color(0, 0, 255));
  } else if (strcmp((char*)payload, "rainbow") == 0) {
    rainbow(10);
  }
  strip.show();

}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  Serial.println("");

  setup_wifi();

  strip.begin();           // INITIALIZE NeoPixel strip object (REQUIRED)

  strip.setBrightness(ledBrightness); // Set BRIGHTNESS to about 1/5 (max = 255)
  for (uint16_t i = 0; i < strip.numPixels(); i++) {
    uint8_t r = 255 * ledBrightness / 255;
    uint8_t g = 200 * ledBrightness / 255;
    uint8_t b = 0 * ledBrightness / 255;
    strip.setPixelColor(i, strip.Color(r, g, b));
    strip.show();
  }

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  client.subscribe("inTopic");
}

boolean reconnect() {
  Serial.println("reconnecting()");
  if (client.connect("arduinoClient")) {
    // Once connected, publish an announcement...
    //client.publish("outTopic","hello world");
    // ... and resubscribe
    client.subscribe("inTopic");
  }
  return client.connected();
}

void loop() {
  if (!client.connected()) {
    long now = millis();
    if (now - lastReconnectAttempt > 5000) {
      lastReconnectAttempt = now;
      // Attempt to reconnect
      if (reconnect()) {
        Serial.println("Connected...");
        lastReconnectAttempt = 0;
      }
    }
  } else {
    // Client connected

    client.loop();
  }

}

// Rainbow cycle along whole strip. Pass delay time (in ms) between frames.
void rainbow(int wait) {
  // Hue of first pixel runs 5 complete loops through the color wheel.
  // Color wheel has a range of 65536 but it's OK if we roll over, so
  // just count from 0 to 5*65536. Adding 256 to firstPixelHue each time
  // means we'll make 5*65536/256 = 1280 passes through this outer loop:
  for (long firstPixelHue = 0; firstPixelHue < 5 * 65536; firstPixelHue += 256) {
    for (int i = 0; i < strip.numPixels(); i++) { // For each pixel in strip...
      // Offset pixel hue by an amount to make one full revolution of the
      // color wheel (range of 65536) along the length of the strip
      // (strip.numPixels() steps):
      int pixelHue = firstPixelHue + (i * 65536L / strip.numPixels());
      // strip.ColorHSV() can take 1 or 3 arguments: a hue (0 to 65535) or
      // optionally add saturation and value (brightness) (each 0 to 255).
      // Here we're using just the single-argument hue variant. The result
      // is passed through strip.gamma32() to provide 'truer' colors
      // before assigning to each pixel:
      strip.setPixelColor(i, strip.gamma32(strip.ColorHSV(pixelHue)));
    }
    strip.show(); // Update strip with new contents
    delay(wait);  // Pause for a moment

  }
}
```

However I'd very much like to use Homie rather than code all the subscriptions and topics by hand, so help with this issue is very much appreciated.

tavalin, 2020-03-23:
I've put together a new sketch which uses the PubSub MQTT client instead, which can successfully change the LED colour when a message is received.  This shows that this is not a fundamental issue due to the basic ESP WiFi connection stealing CPU cycles away from the LED driver - to me this very much points to something under the hood in the `Homie.setup()` and `Homie.loop()` code causing the issue, though I'm not familiar enough with the code to understand what exactly is causing this.

Here's the sketch that connects to the broker and receives messages:
```
#include <Adafruit_NeoPixel.h>
#include <PubSubClient.h>
#include <ESP8266WiFi.h>

WiFiClient espClient;
PubSubClient client(espClient);

const char* ssid = "xxxxx";
const char* password = "xxxxx";
const char* mqtt_server = "192.168.xx.xx";

////////////////////////////////
// LED
////////////////////////////////

#define LED_COUNT 64
#define LED_PIN D3
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

uint8_t ledBrightness = 255;

long lastReconnectAttempt = 0;

void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  payload[length] = '\0';
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.print(" - ");
  Serial.print(length);
  Serial.println();

  String color = String((char*)payload);

  if (color == "off") {
    strip.fill(strip.Color(0, 0, 0));
  } else if (strcmp((char*)payload, "red") == 0) {
    Serial.println("in red");
    strip.fill(strip.Color(255, 0, 0));
  } else if (strcmp((char*)payload, "green") == 0) {
    Serial.println("in green");
    strip.fill(strip.Color(0, 255, 0));
  } else if (strcmp((char*)payload, "blue") == 0) {
    Serial.println("in blue");
    strip.fill(strip.Color(0, 0, 255));
  } else if (strcmp((char*)payload, "rainbow") == 0) {
    rainbow(10);
  }
  strip.show();

}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  Serial.println("");

  setup_wifi();

  strip.begin();           // INITIALIZE NeoPixel strip object (REQUIRED)

  strip.setBrightness(ledBrightness); // Set BRIGHTNESS to about 1/5 (max = 255)
  for (uint16_t i = 0; i < strip.numPixels(); i++) {
    uint8_t r = 255 * ledBrightness / 255;
    uint8_t g = 200 * ledBrightness / 255;
    uint8_t b = 0 * ledBrightness / 255;
    strip.setPixelColor(i, strip.Color(r, g, b));
    strip.show();
  }

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  client.subscribe("inTopic");
}

boolean reconnect() {
  Serial.println("reconnecting()");
  if (client.connect("arduinoClient")) {
    // Once connected, publish an announcement...
    //client.publish("outTopic","hello world");
    // ... and resubscribe
    client.subscribe("inTopic");
  }
  return client.connected();
}

void loop() {
  if (!client.connected()) {
    long now = millis();
    if (now - lastReconnectAttempt > 5000) {
      lastReconnectAttempt = now;
      // Attempt to reconnect
      if (reconnect()) {
        Serial.println("Connected...");
        lastReconnectAttempt = 0;
      }
    }
  } else {
    // Client connected

    client.loop();
  }

}

// Rainbow cycle along whole strip. Pass delay time (in ms) between frames.
void rainbow(int wait) {
  // Hue of first pixel runs 5 complete loops through the color wheel.
  // Color wheel has a range of 65536 but it's OK if we roll over, so
  // just count from 0 to 5*65536. Adding 256 to firstPixelHue each time
  // means we'll make 5*65536/256 = 1280 passes through this outer loop:
  for (long firstPixelHue = 0; firstPixelHue < 5 * 65536; firstPixelHue += 256) {
    for (int i = 0; i < strip.numPixels(); i++) { // For each pixel in strip...
      // Offset pixel hue by an amount to make one full revolution of the
      // color wheel (range of 65536) along the length of the strip
      // (strip.numPixels() steps):
      int pixelHue = firstPixelHue + (i * 65536L / strip.numPixels());
      // strip.ColorHSV() can take 1 or 3 arguments: a hue (0 to 65535) or
      // optionally add saturation and value (brightness) (each 0 to 255).
      // Here we're using just the single-argument hue variant. The result
      // is passed through strip.gamma32() to provide 'truer' colors
      // before assigning to each pixel:
      strip.setPixelColor(i, strip.gamma32(strip.ColorHSV(pixelHue)));
    }
    strip.show(); // Update strip with new contents
    delay(wait);  // Pause for a moment

  }
}
```

However I'd very much like to use Homie rather than code all the subscriptions and topics by hand, so help with this issue is very much appreciated.

kleini, 2020-03-24:
You're using D3 to control your LED strip. Homie uses LED_BUILTIN to drive some LED with blinking for WiFi and MQTT connect: https://homieiot.github.io/homie-esp8266/docs/3.0.0/advanced-usage/built-in-led/. Please check whether your pin D3 collides with LED_BUILTIN. If this is the case, the Homie LED blinking kills your communication to your LED strip.

tavalin, 2020-03-24:
@kleini Really appreciate your persistence with this! :)

I've produced another sketch to test out your theory - it firstly sets the LED pin to something other than D3 and then disables the LED feedback.  Unfortunately it's the same scenario, any attempt to control the LED after the `Homie.setup()` fails.  In the sketch below - the LED sticks on green instead of turning blue.

```
#include <Homie.h>
#include <Adafruit_NeoPixel.h>

#define FW_NAME "test-fw"
#define FW_VERSION "1.0.0"

#define LED_COUNT 1
#define LED_PIN D3
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  strip.begin();
  Serial << "Red..." << endl;
  strip.setPixelColor(0, strip.Color(255, 0, 0)); //RED
  strip.show();

  Serial.println(FW_NAME FW_VERSION);
  Homie_setFirmware(FW_NAME, FW_VERSION);

  Homie.setLedPin(D4, HIGH);
  Homie.disableLedFeedback();
  
  Serial << "Green..." << endl;
  strip.setPixelColor(0, strip.Color(0, 255, 0));  // GREEN
  strip.show();
  
  Homie.setup();

  Serial << "Blue..." << endl;
  strip.setPixelColor(0, strip.Color(0, 0, 255));  // BLUE
  strip.show();

}

void loop() {

}
```

kleini, 2020-03-24:
What about RX and TX? Homie uses output on serial console. Does D3 collide with one of these? Try to use something else as D3 for your LED strip.

tavalin, 2020-03-24:
In my case, the PCB has already been manufactured with the LED on D3 so I can't use another pin (easily).

However I've made some progress on working out what is happening...

I've done some digging into the code and through trial and error (commenting out sections of code) I discovered that the following lines in `BootNormal.cpp` was causing my problem:

```
#if HOMIE_CONFIG
  ResetHandler::Attach();
#endif
```

On my D1 Mini, the reset is on pin D3.  The `ResetHandler::Attach()` is redefining pin D3 as an `INPUT `rather than an `OUTPUT`.

[The documentation](https://homieiot.github.io/homie-esp8266/docs/3.0.0/advanced-usage/resetting/) shows that I can disable the reset functionality by calling `Homie.disableResetTrigger()`.

I have added `Homie.disableResetTrigger()` to my original sketch and have confirmed this now allows me to control the LEDs through the MQTT messages :)

@kleini thank you for your help and suggestions which allowed me to solve my issue

kleini, 2020-03-25:
I am glad, that we found the cause. This needs to be documented somewhere for homie-esp8266, which parts need to be checked, if some hardware does not work as expected as there may be other people stumbling over colliding GPIO usages.

closed 2020-03-25 by kleini