homieiot/homie-esp8266 #674: v3.0.0 docs update needed

mlucjan, 2020-04-30:
Hey!

Firstly, thanks for developing this awesome framework! It was a true game changer for me :).

While developing a home automation system using Homie I have stumbled upon a few errors in the documentation. The most critical one that gave me the most headache while debugging was the MQTT topic name for controlling range properties. As for v.3.0.0 the topic name that is specified in the docs is `homie/<device id>/strip/led_1/set`. Using this topic gives error "Range index 0 is not within the bounds of _HomieNode id here". The topic name that worked for me is actually in the form of: `homie/<device id>/strip_1/led/set`.

Second and not so much critical issue is that the example code for the range properties is using the outdated syntax for both the HomieNode constructor and .advertise method. The code in the attached example .ino sketch is up to date but I just noticed this inconsistency and thought it might be misleading for some people.

The docs page I'm referring to is: https://homieiot.github.io/homie-esp8266/docs/3.0.0/advanced-usage/range-properties/

Thanks!

luebbe, 2020-05-04:
Thanks for reporting. I have never used range properties. Will have to buy a LED strip for testing ;)

codefrey, 2020-05-28:
I think this is more of a bug in Homie and not a documentation issue. Enumerating strips instead of LEDs on a strip does not really make sense.