homieiot/homie-esp8266 #676: D1 mini is crashing after enabling mqtt tls

umairyounus, 2020-05-06:
Hi, I'm using homie-esp8266/examples/LightOnOff/LightOnOff.ino. If I enable TLS on MQTT, it's crashing. Any idea?

`✔ Wi-Fi connected, IP: 192.168.40.26
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...

Exception (9):
epc1=0x40259cf6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0003aaaa depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebf0 end: 3fffffb0 offset: 0190
3fffed80:  00000010 00000010 3fff37a4 4025a03c  
3fffed90:  d0fa686b 1f671c9e eb4e05aa 413e7b10  
`

codefrey, 2020-05-28:
Have you enabled PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY? Maybe the stack is overflowing.

umairyounus, 2020-05-28:
Hi Codefrey,

I've tried it but the same result.

codefrey, 2020-05-28:
That was quick. :) Can you decode the output?

umairyounus, 2020-05-28:
I have been waiting for it :), not sure how to decode it.

codefrey, 2020-05-28:
Are you using Arduino-IDE or PlatformIO?

umairyounus, 2020-05-28:
PlatformIO

codefrey, 2020-05-28:
Okay, then take a look here on how to setup a filter to decode the messages: https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters

umairyounus, 2020-05-28:
Ok Markus, I'll try it shortly. Cheers

umairyounus, 2020-05-29:
Sorry for the delay, here are the results:

Decoding stack results
0x4025f3ec: basic_read at ssl/tls1.c line 1607
0x4025500f: tcp_receive at core/tcp_in.c line 1298
0x40256544: tcp_input at core/tcp_in.c line 501
0x4025c11e: ip4_input at core/ipv4/ip4.c line 1467
0x40252fd5: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40252df4: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x40283a1a: ethernet_input at glue-esp/lwip-esp.c line 365
0x40283a2b: ethernet_input at glue-esp/lwip-esp.c line 373
0x40237c1c: strtol at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/strtol.c line 224
0x40241085: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48

codefrey, 2020-05-30:
Thanks for providing the decoded trace. I think I'll have to pass at this point. For me it looks like the issue is not directly caused by Homie but either by lower level code or by running out of memory. My bet would be on the latter.