homieiot/homie-esp8266 #678: Build issue with current stable version (v3) via PlatformIO in ESPAsyncWebServer.h due to missing ESP8266WiFi.h

deradam, 2020-05-19:
Seems like the issue #465 is back again. I today installed VSC, Platform.io on a fresh system and created a new project. platform.ini is
~~~
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino

lib_deps = Homie
~~~
And I am getting the same error as described above.
~~~
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.5.1 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20701.0 (2.7.1) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Homie library in registry
Found: https://platformio.org/lib/show/555/Homie
LibraryManager: Installing id=555
Homie @ 3.0.0 is already installed
Installing dependencies
LibraryManager: Installing id=64 @ ^6.11.4
ArduinoJson @ 6.15.2 is already installed
Looking for AsyncMqttClient library in registry
Found: https://platformio.org/lib/show/346/AsyncMqttClient
LibraryManager: Installing id=346 @ ^0.8.0
AsyncMqttClient @ 0.8.2 is already installed
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: https://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ 1.2.0
ESPAsyncTCP @ 1.2.0 has been successfully installed!
Looking for AsyncTCP library in registry
Found: https://platformio.org/lib/show/1826/AsyncTCP
LibraryManager: Installing id=1826 @ ^1.0.0
AsyncTCP @ 1.1.1 is already installed
Looking for Bounce2 library in registry
Found: https://platformio.org/lib/show/1106/Bounce2
LibraryManager: Installing id=1106 @ ^2.1.0
Bounce2 @ 2.53 has been successfully installed!
Looking for ESP Async WebServer library in registry
Found: https://platformio.org/lib/show/306/ESP Async WebServer
LibraryManager: Installing id=306

Downloading  [##########################----------]   72%
Downloading  [###########################---------]   75%
Downloading  [############################--------]   78%
Downloading  [#############################-------]   81%
Downloading  [##############################------]   84%
Downloading  [###############################-----]   87%
Downloading  [################################----]   90%
Downloading  [#################################---]   93%
Downloading  [##################################--]   96%
Downloading  [####################################]  100%
ESP Async WebServer @ 1.2.3 has been successfully installed!
Installing dependencies
LibraryManager: Installing id=305
ESPAsyncTCP @ 1.2.0 is already installed
LibraryManager: Installing id=1826
AsyncTCP @ 1.1.1 is already installed
Looking for Hash library in registry
Warning! Library `{'name': 'Hash', 'platforms': ['espressif8266']}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'Hash', 'platforms': ['espressif8266']}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Homie> 3.0.0
|   |-- <ArduinoJson> 6.15.2
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |-- <Bounce2> 2.53
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson> 6.15.2
|   |-- <ESPAsyncTCP> 1.2.0
Building in release mode
Compiling .pio/build/d1_mini/src/main.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Archiving .pio/build/d1_mini/lib199/libArduinoJson_ID64.a
Compiling .pio/build/d1_mini/liba81/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
Compiling .pio/build/d1_mini/liba81/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
Compiling .pio/build/d1_mini/liba81/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Compiling .pio/build/d1_mini/liba81/ESPAsyncTCP_ID305/SyncClient.cpp.o
Compiling .pio/build/d1_mini/liba81/ESPAsyncTCP_ID305/tcp_axtls.c.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PingRespPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubAckPacket.cpp.o
Indexing .pio/build/d1_mini/lib199/libArduinoJson_ID64.a
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubCompPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRecPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRelPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PublishPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/SubAckPacket.cpp.o
Compiling .pio/build/d1_mini/lib212/AsyncMqttClient_ID346/AsyncMqttClient/Packets/UnsubAckPacket.cpp.o
Compiling .pio/build/d1_mini/lib42b/Bounce2_ID1106/Bounce2.cpp.o
Compiling .pio/build/d1_mini/lib0c0/Hash/Hash.cpp.o
Compiling .pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncEventSource.cpp.o
Compiling .pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncWebSocket.cpp.o
In file included from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncEventSource.h:29:0,
                 from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncEventSource.cpp:21:
.pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
*** [.pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncEventSource.cpp.o] Error 1
In file included from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncWebSocket.h:32:0,
                 from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncWebSocket.cpp:22:
.pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
*** [.pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncWebSocket.cpp.o] Error 1
============================================== [FAILED] Took 13.99 seconds ==============================================
The terminal process terminated with exit code: 1
~~~
Any known issues here? Thanks in advance!

deradam, 2020-05-21:
After some reading and fiddeling around with the build, I finally made it compile with the following config file.

~~~
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY

lib_deps =
  ESP8266WiFi
  ESP8266mDNS
  ESP8266HTTPClient
  DNSServer
  WiFi
  SPI
  Ticker
  Homie
~~~

Maybe it helps somebody else.

deradam, 2020-06-01:
Okay, I think I have taken it too quick as being resolved. While loading it onto the Wimos, I have the following errors:
~~~
 /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware homie-minimum (0.0.1)
🔌 Booting into config mode 🔌
Device ID is a020a61a81f2
AP started as Homie-a020a61a81f2 with IP 192.168.123.1

Exception (9):
epc1=0x40229bde epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000049 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  3ffef9d0 3ffffec0 3ffef9d0 4020e6f4  
3ffffeb0:  3ffef9d0 3ffef9b8 3ffef9d0 40220ca3  
3ffffec0:  2e777700 3fffff00 80fef9d0 4020ea80  
3ffffed0:  3ffef9d0 3ffeff00 3ffef9b8 40220d04  
3ffffee0:  00000000 3ffef9b8 3ffef8d0 4020498f  
3ffffef0:  696d6f48 30612d65 36613032 31386131  
3fffff00:  00003266 3fffff90 3fffff20 3fff02d0  
3fffff10:  4022b088 0000002a 017ba8c0 8120d4e0  
3fffff20:  4022c680 40220035 017ba8c0 4022b0a0  
3fffff30:  4022b088 4022c6e8 00ffffff 4020d8f5  
3fffff40:  00000000 3ffefa38 3ffef800 3fff02d0  
3fffff50:  00000000 3ffeff78 3ffef800 4020172d  
3fffff60:  3fff25e4 00000000 40201348 4022abb8  
3fffff70:  402010a0 3fff01a0 3ffef800 3fff02d0  
3fffff80:  3fffdad0 00000000 3ffef800 402011fd  
3fffff90:  40201034 feefeffe 4022ab80 4022ab6c  
3fffffa0:  feefeffe feefeffe 3fff0290 4020fb4c  
3fffffb0:  feefeffe feefeffe 3ffe864c 40100f91  
<<<stack<<<
~~~

Ideas anyone? Thanks in advance!

snikch, 2020-07-16:
@deradam did you ever get this resolved? Running into the exact same issues, both the setup then the same crash.

snikch, 2020-07-16:
Decoded, this is what I'm getting.

```
Exception Cause: 9  [LoadStoreAlignmentCause: Load or store to an unaligned address]

0x40229aca: UdpContext::unref() at ??:?
0x4020e5e0: String::~String() at ??:?
0x40220b8f: DNSServer::downcaseAndRemoveWwwPrefix(String&) at ??:?
0x4020e96c: String::operator=(String const&) at ??:?
0x40220bf0: DNSServer::start(unsigned short const&, String const&, IPAddress const&) at ??:?
0x40204aeb: HomieInternals::BootConfig::setup() at ??:?
0x4022af28: HomieInternals::Logger::write(unsigned char const*, unsigned int) at ??:?
0x4022c548: AsyncServer::setNoDelay(bool) at ??:?
0x40220035: HTTPClient::returnError(int) at ??:?
0x4022af40: HomieInternals::Logger::write(unsigned char const*, unsigned int) at ??:?
0x4022af28: HomieInternals::Logger::write(unsigned char const*, unsigned int) at ??:?
0x4022c5b0: AsyncServer::setNoDelay(bool) at ??:?
0x4020d7e1: Print::write(char const*) at ??:?
0x402018f9: HomieInternals::HomieClass::setup() at ??:?
0x402010ac: delayHandler(HomieRange const&, String const&) at ??:?
0x4022aa2c: std::_Function_handler<bool (HomieRange const&, String const&), bool (*)(HomieRange const&, String const&)>::_M_invoke(std::_Any_data const&, HomieRange const&, String const&) at ??:?
0x4022aa44: std::_Function_base::_Base_manager<bool (*)(HomieRange const&, String const&)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40201301: setup at ??:?
0x402010ac: delayHandler(HomieRange const&, String const&) at ??:?
0x4022aa44: std::_Function_base::_Base_manager<bool (*)(HomieRange const&, String const&)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x4022aa2c: std::_Function_handler<bool (HomieRange const&, String const&), bool (*)(HomieRange const&, String const&)>::_M_invoke(std::_Any_data const&, HomieRange const&, String const&) at ??:?
0x4020fa38: loop_wrapper() at core_esp8266_main.cpp:?
0x40100f91: cont_wrapper at ??:?
0x4010f000: ?? ??:0
```

luebbe, 2020-07-16:
I'm using homie 3 stable a lot without boot crashes and I only need to add extra libraries (BME280 and so on) to the lib_deps. I never used the LWIP2_LOW_MEMORY define. Could you try without that?
@snikch If you find out how to get the line numbers in the decoded stack trace, please let me know :D

snikch, 2020-07-16:
@luebbe What device are you using? I use a variety with older homie versions but specifically am using a sonoff basic at the moment.

luebbe, 2020-07-16:
A handful of Sonoff basic and sonoff S20, lots of D1 Mini one D1 Mini pro and some bare off-the-shelf ESP12F soldered onto prototype boards, which I bought at the beginning before I knew how good the D1 Minis are.
You can take a look at the platformio.ini of my different projects: https://github.com/luebbe?tab=repositories.
There are some issues caused by AsyncMqttClient (which I didn't work around), but since a few days there's a remedy called https://github.com/philbowles/PangolinMQTT.
I hope to find time "soon" to port Homie to Pangolin.

snikch, 2020-07-16:
Hopefully I'll get a chance to have a look over the weekend. In the end I just went back to `2.0.0` to get my device working (duty cycle relay for a hydroponics pump).

larryn46, 2022-06-14:
In regards to the previous post, I found user_interface.h in ~/Arduino/Arduino-master/tools/sdk/include/. I copied user_interface.h  to this project's include file. When I compiled, I got an error message: (I'm leaving out most of the compiler message)
In file included from .pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:22,
                 from src/main.cpp:2:
include/user_interface.h:28:10: fatal error: os_type.h: No such file or directory

If you go to the user_interface.h file and look at line 28,  sure enough, there is an include to this h file.  In fact, there are a total of 6 includes. 

For the programmers that love PlatformIO, there must be a shorter way to handle these problems. Do any of you know what that method is?

larryn46, 2022-06-14:
I tried adding the following to platform.ini 
lib_extra_dirs = 
  /home/fido/Arduino/libraries
  /home/fido/Arduino/Arduino-master/tools/sdk/include
The latter is the directory in which user_interface.h, os_type.h, and the other requested files are found. The simple program still wouldn't compile. The same error message occurred, (which i am leaving out most of it, just the error part):
In file included from .pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:22,
                 from src/main.cpp:2:
include/user_interface.h:28:10: fatal error: os_type.h: No such file or directory
Does anyone know how to solve this problem?

larryn46, 2022-06-15:
I tried platformio.ini with lib_dir instead of lib_extra_dirs =
The entire platformio.ini is shown here:
    [env:esp32doit-devkit-v1]
    platform = espressif32
    board = esp32doit-devkit-v1
    framework = arduino
    lib_dir = 
      /home/fido/Arduino/libraries
      /home/fido/Arduino/Arduino-master/tools/sdk/include
    lib_ldf_mode = chain+
    lib_deps =
      tzapu/WiFiManager
It seems like lib_dir doesn't work. Any ideas?

trysten, 2023-07-08:
I'm confused. Nothing worked until I removed `lib_deps` from `platformio.ini`, then it automatically detects homie and works :sob:

NicoDiercks, 2025-10-23:
I am not sure, if I had the same issue, but PlatformIO was not able to compile Homie due to a dependency issue. 
I had to edit the library.json of Homie to fix it.
See #747