homieiot/homie-esp8266 #680: settings section not created by ui_bundle

mosseb, 2020-06-03:
Hi.
I have 2 custom settings like that :
```
upCourseTimeSetting.setDefaultValue(SHUTTER_COURSETIME_MAX).setValidator(validator);
downCourseTimeSetting.setDefaultValue(SHUTTER_COURSETIME_MAX).setValidator(validator);
```
Initially I manually add an empty `settings` section in the JSON config that I upload via SPIFFS.
Then I use MQTT message to populate the values.
So far so good.

In the case the configuration goes in an unwanted state (wrong manipulation), I implemented a way to reset the module to default settings, by issuing a Homie.reset();
It restarts, goes into configuration mode and spawns the AP.
I enter all the needed informations, included values for the custom settings, it reboots...
And unfortunately, the settings `section` is lost and doesn't exist anymore in `$implementation/config`

And I cannot add it again because of the error :
`✖ Config does not contain a settings object`

I would expect the both things to be possible :
- I can add a custom settings to the JSON settings by passing it on the `$implementation/config/set` topic
- ui bundle add the section automatically when detecting some custom settings
- Or maybe the simplest solution would be to always add an empty `settings` section at boot if it does not exist, or remove the "config does not contain a settings object" error and actually add it instead.

Please tell me if I miss something to adress my case, but in this state, the module is not able to go to the unaccessible ceiling..

Thanks !

codefrey, 2020-06-04:
A fix for this patching issue is currently included in the develop branch, so try to use this if feasible.

elbowz, 2021-09-02:
Confirm, It works on develop branch :+1: 

Thanks!