homieiot/homie-esp8266 #681: Disable WLAN-Config reading

GefaketHD, 2020-06-14:
Hey,
I like this project a lot. But I have one thing, because I am thinking about using this in my own project(s): It should be possible to disable the WLAN-Settings read by the config-file. For example to do Wifi-handling with libs like WiFiMulti. They come with internal credentials handling and therefore I don‘t need to provide creds in the config.json.
As far as I understood the documentation, I HAVE to provide wifi-creds in the config.
Correct me if I‘m wrong.
BR

MajorTwip, 2020-06-15:
You don't normally provide a config.json.

When your ESP starts with a HomieESP-Project on it, it looks on it's filesystem for a config.json

If there is no config.json it starts as an AccessPoint with a Webportal (or at least the API) to configure the credentials. Homie then creates the config.json on its own.

If you like to completely decouple Homie-ESP from the WiFi-Management, this will be much more complicated.

kleini, 2020-06-15:
This is not completely correct. I introduced compiler flags to remove the configuration mode completely from homie-esp8266. My opinion was, that this code is just needed once and then never again and it consumes a lot of space in the flash. See compiler flags for more information: https://homieiot.github.io/homie-esp8266/docs/3.0.0/advanced-usage/compiler-flags/
So I need to upload the WiFi credentials before starting the Homie device into SPIFFS.

Maybe it is possible by introducing another compiler flag to replace reading the WiFi credentials from a library like WiFiMulti. But it will require some coding work.

luebbe, 2020-06-15:
Yes, multiple access points FTW :-)

cafe01, 2020-07-31:
Why does config stuff takes so much flash space? 
It's the reason I can't use OTA update on my (1mb flash) sonoff. 
It fits if I compile with `HOMIE_CONFIG=0` flag.

luebbe, 2020-07-31:
> 
> 
> Why does config stuff takes so much flash space?
> It's the reason I can't use OTA update on my (1mb flash) sonoff.
> It fits if I compile with `HOMIE_CONFIG=0` flag.

That is a misconception. The config is always there. What takes up so much space is the ESP web server, which is 99% of the time only used to create an initial config via the web gui and never used again. If you compile with HOMIE_CONFIG=0, the web server is not included.