homieiot/homie-esp8266 #685: how to use setDefaultValue?

RootShell-coder, 2020-09-30:
I took the code from close issue: #655 and this works great. fills in the field with the default value in web ui_bundle.
I am trying to do the same but for the temperature sensor.

```c++
#include "ds18b20.h"

HomieSetting<long> SensorDS18B20::sensorsInterval("sensorsInterval", "Sensors temperature reading interval in ms (range:5000-600000 default:10000)");
SensorDS18B20::SensorDS18B20(const char* id, const char* name, const char* type) : HomieNode(id, name, type),
    lastSensorLoop(0){
        sensorsInterval.setDefaultValue(10000).setValidator([] (long candidateSensor) {
		    return (candidateSensor > 5000 && candidateSensor < 600000);
	});
}

void SensorDS18B20::setup(){
    advertise("sensor").setName("Temperature").setRetained(true).setUnit("°C").setDatatype("float").setFormat("json");;
}

void SensorDS18B20::loop(){
      if (millis() - lastSensorLoop >= (unsigned long) sensorsInterval.get() || lastSensorLoop == 0) {
        lastSensorLoop = millis();
      }
}
```

but the sensorsInterval field remains empty.

Although at the same time this code perfectly fills the pingInterval field with the default value.
```c++
#include "ping.h"

HomieSetting<long> TestMQTTserver::pingInterval("pingInterval", "Execute ping interval in ms (range:10000-600000 default:60000)");
TestMQTTserver::TestMQTTserver(const char* id, const char* name, const char* type) : HomieNode(id, name, type),
    lastPingLoop(0){
        pingInterval.setDefaultValue(60000).setValidator([] (long candidatePing) {
            return (candidatePing > 10000 && candidatePing < 600000);
    });
}

/* bool TestMQTTserver::handleInput(const HomieRange& range, const String& property, const String& value){
  float time = (float)(micros() - pingTime)/1000;
  Homie.getLogger() << "  • " << property << " from mqtt = "<< time << " ms" << endl;
  setProperty("ping").send(String(time));
  return true;
}
*/
void TestMQTTserver::setup(){
    advertise("check").setName("Ping").setRetained(true).setUnit("ms").setDatatype("float").settable();
}

void TestMQTTserver::loop(){
    if (millis() - lastPingLoop >= (unsigned long) pingInterval.get() || lastPingLoop == 0) {
        lastPingLoop = millis();
        setProperty("ping/set").send("ping");
        pingTime = micros();
    }
}
```
I'm at a loss, what magic was used for ping?
```c++
//this my main.cpp 
#include <Homie.h>
#include "ds18b20.h"
#include "ping.h"

#define BRAND     "THC"
#define FM_NAME   "The Heating Control"
#define FM_VER    "0.0.1"

SensorDS18B20 SensorDS18B20("sensor", "Temperature", "DS18B20");
TestMQTTserver TestMQTTserver("check", "Ping", "PingMQTT");

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  Homie_setBrand(BRAND);
  Homie_setFirmware(FM_NAME, FM_VER);
  Homie.setResetTrigger(1, LOW, 10000);
  Homie.setup();
}

void loop() {
  Homie.loop();
}

```
Sorry if this question has already been asked. Thank you

luebbe, 2020-09-30:
I never set the default values in the constructor. If the default value is missing in homie.setup, homie goes into config mode.

I always implement a method `beforeHomieSetup()` which does stuff like this and which is called in setup after the node's constructor and before homie.setup.

see: 
- https://github.com/luebbe/homie-node-collection/blob/master/examples/demo-sensor-nodes.cpp
- https://github.com/luebbe/homie-node-collection/blob/master/src/AdcNode.cpp

RootShell-coder, 2020-09-30:
exactly, I can see that this is a constructor. Need more rest. sorry.

closed 2020-09-30 by RootShell-coder