homieiot/homie-esp8266 #688: OTA via HTTP/HTTPs

bilogic, 2020-10-19:
Does OTA to a HTTP server still works?
I can't find a server and path in the config of the latest code.

luebbe, 2020-10-20:
As far as I know OTA only works via MQTT. There never was HTTP support.

bilogic, 2020-10-20:
Thanks, I believe there was back in 1.5.0, I had it in my code but recently had to make some changes to my device.

Could you explain how OTA via MQTT works?
I mean, where should I host the firmware so that clients can pull it? I'm using aedes inside nodered.
Or is triggering from the broker the only way?

Thank you.

luebbe, 2020-10-20:
That was well before I started using homie...

The firmware is pushed to the corresponding MQTT topic for a device. There's a python script for it in the homie repository. The device never pulls the firmware in on its own. You could implement something however.

You can also take a look at these projects:
https://github.com/jpmens/homie-ota (small web server)
https://github.com/rttools/hodmin (console)

They both allow you to detect if a device's firmware is up to date and trigger an update.

bilogic, 2020-10-20:
Ok thanks, maybe it is better this way after all.

closed 2020-10-20 by bilogic