homieiot/homie-esp8266 #694: Proposal: A new light weight config mode

kartom, 2020-11-15:
The new  `HOMIE_CONFIG=0 flag` offer the opportunity to completely remove the ESP webserver and save space. The drawback of this is that the device has to be reflashed to change the configuration if WiFi/MQTT connection fails due to a config misstake or that something has been altered in any of those (new wifi password for example).  

My idea of configuration mode goes something like this: 
1. Start WiFi in AP-mode.
2. Wait for the first client to connect
3. Connect to an open MQTT broker on that client
4. Wait for an new configuration sent over MQTT.

By writing an Android app that contains an MQTT broker and a configuration interface it will be possible to access the configuration w.o. the need for much extra code. (Quick investigations gave that the AP-mode was rather lightweight).
My first test with Moquette as broker on Android was promising.

Does anybody think this would be useful?

luebbe, 2020-11-16:
This would be very useful since it would even work without a companion app. E.g. any old netbook with lubuntu would do.

But:
The http json api is always there in config mode, so you can always send a config via `curl -X PUT`. So in princible your suggestion is not very different from what is possible now. I think the real problem is that homie never falls back to a config/AP mode if Wifi/MQTT connections fail for a while.

Would the following work?

1. Try to connect to wifi/mqtt with the currently know credentials
2. After N failures, open an AP so that a new configuration can be sent.
3. If nobody has connected to the AP (or no new config was received) for N minutes go back to 1.

kartom, 2020-11-16:
If i use the `HOMIE_CONFIG=0`  and put that code onto a ESP8266 without a config file i get the following response:
```
✖ /homie/config.json doesn't exist
Configuration invalid. CONFIG MODE is disabled.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```
And then the device reboots and repeats itself so it doesn't seam that the http json api is available if i select `HOMIE_CONFIG=0`. 

To get an Homie device into configuration mode could be solved by using `Homie.reset()` or `Homie.setResetTrigger()`. (In my case i am using the Homie.setResetTrigger() on one of the physical buttons that are already connected to the device and used for other purposes as well. This way i can reset it and make it ready to accept a new configuration without the need for reprogramming of the device. It hasn't been used yet, but the day i'll have to change the WiFi password i think i will be most grateful for the feature.

But the original question and idea remains valid, would it be useful for anybody more than me to have this functionality described in the proposal.

luebbe, 2020-11-17:
Yes, you're right, when the webserver is gone, there's no http api :)
And yes, I think your proposal sounds helpful. I always compile my projects without webserver.

luebbe, 2020-11-22:
A lightweight config mode would have come handy for me already. Yesterday I flashed one of my devices using HOMIE_CONFIG=0 and didn't pay attention to the memory layout. Of course I lost the existing config :) 
How is your progress?

Maenthyx, 2020-12-26:
Hi,
I really enjoy homie-esp8266 since some days and would love to see this project to evolve. I wonder whether there are some news to this topic? May I add my2cents?
In my opinion, conversion of any existing filesystem/data on running systems is out-of-scope of this project. Adding code to support an automatic conversion from spiffs to littlefs (whatever this might look like) would result in (a lot of additional) code/dependencies which doesn't fit into the "Homie" scope. Have in mind that this would raise future efforts to maintain that "legacy code".
I also couldn't find any other project considering a conversion from spiffs to littlefs on running systems, so why should this project do?

Nevertheless, I feel the pain loosing data. So I'd suggest a process (and maybe tools/simple sketches) to download the data, update the system and upload the data. Either manually done, or by leveraging the OTA feature. Well documented, but not implemented into this project.

You also might want to read this thread, discussing this topic, options and efforts.
Start reading from "Feb 25 14:14": [Gitter Thread](https://gitter.im/esp8266/Arduino?at=5e551db3101adb4160c680f1)

  
  
Also, for a limited time and depending on the effort, both spiffs and littlefs could be supported by using compiler flags, resulting in running either spiffs or littlefs (but not both at the same time). So people have some time to migrate their systems.

Thoughts?

kleini, 2020-12-27:
If you compile with `HOMIE_COMFIG=0`, then there is no config mode compiled into the code. Furthermore in your case (@kartom) JSON configuration is missing. The device is then not able to open the access point and webserver, as this code is not compiled in and it does not have a configuration to connect to any access point. The only solution, that I found for this case, where the device is completely stuck, is to write the log message and do a reboot. Another possible option is to halt the device and do nothing any more.

kartom, 2020-12-31:
The problem with HOMIE_CONFIG=0 is that there is no possible way to access the configuration of the device (other than direct flash). 
My progress is slowly moving forward since a lot of other thins had required my attention, but my ambition is still to implement this idea.