homieiot/homie-esp8266 #702: Cycle reload Exception (9) after AP start

Denages, 2021-01-03:
I've built several projects using Homie. But for now, I have a weird trouble.
I use Platform.io and esp8266 (esp12 1mb) module. It's connected to a separate DC power adapter.
After flashing device goes into an infinity loop. What could be the reason?

Code:
```
#include <Homie.h>

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial << endl << endl;
  Homie_setFirmware("bare-minimum", "1.0.0"); // The underscore is not a typo! See Magic bytes
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

Log:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
STARTED

✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware wc_indicator (1.0.0)
🔌 Booting into config mode 🔌
Device ID is 18fe34fafb83
AP started as WC_Indicator-18fe34fafb83 with IP 192.168.123.1

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4022944e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000049 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  3ffef9c8 3ffffed0 3ffef9c8 4020de54  
3ffffec0:  3ffef9c8 3ffef9b0 3ffef9c8 40222c73  
3ffffed0:  2e777700 3fffff00 80fef9c8 4020e1e0  
3ffffee0:  3ffef9c8 3ffefef0 3ffef9b0 40222cd4  
3ffffef0:  00000000 3ffef9b0 3ffef8c8 402047aa  
3fffff00:  495f4357 6369646e 726f7461 6638312d  
3fffff10:  66343365 38626661 3fff0033 3fff02d0  
3fffff20:  4022a560 0000002a 017ba8c0 8120cc38  
3fffff30:  4022bb38 40220035 017ba8c0 4022a578  
3fffff40:  4022a560 4022bba0 00ffffff 4020d04d  
3fffff50:  00000000 3ffefa30 3ffef7f8 3fff02d0  
3fffff60:  00000000 3ffeff70 3ffef7f8 40201679  
3fffff70:  4022a560 3ffefa30 3ffe88df 4020142c  
3fffff80:  3fffdad0 3ffefa30 3ffef7f8 3fff02d0  
3fffff90:  3fffdad0 3ffefa30 3ffef7f8 402010bd  
3fffffa0:  feefeffe 00000000 3fff0290 4020f2ac  
3fffffb0:  feefeffe feefeffe 3ffe864c 40100f71  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
...
```

Decoder says:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4022944e
EXCVADDR: 0x00000049

```

luebbe, 2021-01-03:
You can add a monitor_filter to your pio config: https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#filters
This saves you the extra decoder step.

First I'd replace the `board=esp12e` with `board = esp01_1m`. This is probably the reason, because by default an esp_01 is assumed to have 4mb of flash.

Did you power cycle the device? Also I've read in several places that some devices need the flash to be erased completely before you can re-flash them.

Denages, 2021-01-03:
@luebbe 
I've changed board to esp01_1m. The output became:
```

✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
�
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
�
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

kartom, 2021-01-03:
Are you sure that your power supply has enough capacity for the current spikes from the ESP? 
Have you used that supply for other ESP projects? Otherwise i suggest you try to put a capacitor >=22uF on the power to see if that helps.