homieiot/homie-esp8266 #704: My router shows the incorrect device name.

davida72, 2021-01-05:
New to PlatformIO and Homie, so this could be user error. I'm working my way through the Getting Started guide and the example below is [LightOnOff relay](https://github.com/homieiot/homie-esp8266/blob/develop/examples/LightOnOff/LightOnOff.ino).

I have a D1 Mini and I'm using this setup.

```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 115200
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
lib_deps = 
    Homie
```

_Quick aside, it would have been great if the guide had explained to set monitor_speed and then use the debug function. It's taken me a few hours to get here!_

I'm able to set up the device in configuration mode, whereupon it reboots. What confused me is the fact that it's not using the device name when connecting to the wifi and my router is still using the device name from when it was using an WLED firmware. Here's the debug output.

```
💡 Firmware awesome-relay (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: 483fda78b667
  • Device ID: homieamor
  • Name: HomieAmor
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: homenetwork
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.86.73
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: mqtt
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.86.52
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Interval: 65s (60s including 5s grace time)
  • Wi-Fi signal quality: 96%
  • Uptime: 7s
```

At this point I was expecting my router (Google Wifi) to be showing a device called homieamor and perhaps even my MQTT broker picking up a new device. Am I doing something wrong or is this a bug? Also, is there a discord for ESP8266 Homie?

luebbe, 2021-01-05:
It's the same with my router. It keeps the first device name that is assigned to a MAC address. You have to delete it and reconnect your device. Even that didn't work every time for me. I don't think it's a problem with homie

davida72, 2021-01-05:
Interesting. Thank you.

I'd have thought there would be more people talking about this problem on the internet if new device names didn't replace old ones on routers. I have the ability to rename it within my router so I guess it isn't a big deal.

I'll close this in a day or so if nobody else has anything to say.

closed 2021-01-09 by davida72