homieiot/homie-esp8266 #705: Presentation: Single-File solution for Homie OTA

MajorTwip, 2021-01-21:
Hi

I made a small Webpage (all in one file) to update Homie3 devices.
Just download the html and open it in a browser.

It supports: 

- overview of the Devices
- OTA-Update (with Magic-Bytes detection)
- Config update

Feel free to enhance/copy

https://github.com/MajorTwip/HomieControl

kleini, 2021-01-21:
This looks promising. How can I provide credentials for my MQTT broker?
What is the advantage of this project? I was already very satisfied with https://github.com/skoona/HomieMonitor

MajorTwip, 2021-01-31:
Hi
Credentials not yet implemented as I dont use them.
Advantage is solely the fact that there is no installation needed. Only prerequisite is a Browser.

kleini, 2021-02-02:
Installing a single web page or running a container is today not a very big difference in setup efforts.

RootShell-coder, 2021-12-02:
Very interesting, but you need credentials to connect to the broker