homieiot/homie-esp8266 #706: Conditional HomieNode creation

SvenDowideit, 2021-03-07:
I would like to have one firmware that has code for all the sensors that I use, and then to use the configuration settings to decide which to enable (d1 wemos, and non-technical users)

```
SensorNode *sensors[10];
uint sensorCount = 0;

HomieSetting<bool> hasButton("button", "has button"); // id, description

void setup()
{
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie_setBrand(FW_BRAND);

  hasButton.setDefaultValue(false);

  Homie.setup();
  if (hasButton.get())
  {
    Serial.printf("adding a button");
    sensors[sensorCount++] = new ButtonNode("button", "test", D3);
  }
}
```

but atm, it looks like the settings aren't parsed until `Homie.setup()` is called, and the devices need to be created before `Homie.setup()` is called.

is there a right way to do this? or should I look at making a PR to separate out the configuration parsing and the setup?

luebbe, 2021-03-07:
Take a look at my [homie node collection](https://github.com/luebbe/homie-node-collection/tree/master/src). There are some examples (e.g. the bme280 node) where the (per node) settings are created at run time. I don't creat the nodes at run time, because  I don't have your use case, but maybe it can help you to get started.

SvenDowideit, 2021-04-01:
@luebbe would you be interested in a PR (or just do it - I don't know enough about testing forks of platformio modules yet) to add a `HomieClass::LoadConfig()` that can be called before the `Homie.setup()`?

Admittedly, I need to work out how to do it soon, as I think I want to take that further :)

luebbe, 2021-04-01:
Sure, go ahead. But I'm not that deep into the homie sources to provide a good review, depending on the complexitiy of the PR.

SvenDowideit, 2021-04-01:
well darn. I have something that works for me, but is clearly naive.

but at the same time, Platformio (or a vscode plugin) has reformatted so much code as to be a useless diff.

guess this will take a few weeks (assuming we get to go on the no-computer easter holiday)

I've got some ideas tho :)

RunningPenguin, 2021-05-26:
Hi, is here any progress?

I have a similar use case. I would like to provide a firmware which is able to connect to 1-5 sensors over serial interface. Each sensor has to be paired on its own and has its own serial number.

To achieve this my plan was to use five HomieSetting<const char *> for which the end user could write his serial number(s) and leave some blank if he has less than 5 sensors attached. For example he provides the information's with the Android "Homie setup" App.
Then I would like to count the amount of given serial numbers and either create an individual node (at runtime) or create/advertise the corresponding amount of values (at runtime).

I have to put the serial numbers in a char array too, that is right now a problem I haven't resolved yet.

I had already a look into the BME280 Node from @luebbe and in the documentation of the HomieSetting but right now I have no plan if the use-case is possible or not without changes to the Homie esp8266 implementation. Perhaps somebody could give me a few tips?

Thanks in advance