homieiot/homie-esp8266 #708: add logging via MQTT

CWempe, 2021-05-07:
When developing my projects I do not use a serial or USB cable, but push the firmware via MQTT to my device.

For that reason it is not possible to see the log messages created by `Homie.getLogger()`.
I think it would be a great improvement if this was possible.

I would propose to add an option where Homie sends all `getLogger` messaged via mqtt, too.

Like: `homie/devicename/$logger "This is an important debug message`"

Ideally this could be turned on/off via mqtt.
This way you could even debug devices that are installed in hard to reach places.

Instead of using the `getLogger` we could add a new function, that could support severity level.

Like:

```c
Homie.mqttLogger.info() << "This is an info message!" << endl;
Homie.mqttLogger.debug() << "This is a debug message!" << endl;
Homie.mqttLogger.error() << "This is an error message!" << endl;
```

Result:

```
homie/devicename/$logger/info "This is an info message!"
homie/devicename/$logger/debug "This is a debug message!"
homie/devicename/$logger/error "This is an error message!"
```

Another option would be to support syslog, but I think mqtt would be fine for most users. :)

CWempe, 2021-05-09:
This is great!
Thanks 👍 

I think it should be listed in the ["Community Projects" section](https://homieiot.github.io/homie-esp8266/docs/stable/others/community-projects/).

luebbe, 2021-05-10:
@euphi has got some great stuff in his repositories. I came across his LoggerNode when I started using homie and thought "this may be helpful for me some day" and have never used it yet. Now it is at least helpful for you, because I remember it.

A PR is always welcome ;)

closed 2021-05-11 by CWempe