homieiot/homie-esp8266 #711: Homie isn't compatible with Espressif 8266 platform core v3.0.0

RunningPenguin, 2021-06-05:
Hi,

I just updated my platform Espressif 8266 in PlatformIO from version 2.5.0 to version 3.0.0

[image]

Then some projects which are working with the "old" version stopped to work.

The first issue is these one:
`.pio/libdeps/d1_mini/Homie/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_proxyHttpRequest(AsyncWebServerRequest*)':`
`.pio/libdeps/d1_mini/Homie/src/Homie/Boot/BootConfig.cpp:315:20: error: call to 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)`
`  315 |   _httpClient.begin(url);`
`      |   ~~~~~~~~~~~~~~~~~^~~~~`

I haven't investigated if there are more issues. Perhaps the fix is just to add three lines like described here:
[https://community.platformio.org/t/bool-httpclient-begin-string-is-deprecated/16924](url)

But for the first the addition of the platform version in platformio.ini is a workaround

`platform = espressif8266 @ 2.5.0`

unrelentingtech, 2021-06-12:
> add three lines like described here

Kinda, the fix is to add `WiFiClient _wifiClient;` to the class in `BootConfig.hpp` and then fix the call (`_httpClient.begin(_wifiClient, url);`)

closed 2022-03-01 by luebbe

mrRo8o7, 2022-04-30:
@unrelentingtech Thank you!