homieiot/homie-esp8266 #715: Problem with range-properties in Homie 3

thecky, 2021-07-26:
Hi,

maybe I miss understood something with the "new" range-advertisment, but currently it doesn't work for me.
Even the example LedStrip (https://github.com/homieiot/homie-esp8266/blob/v3.0/examples/LedStrip/LedStrip.ino) doesn't work as I expected.

I also assume that the documentation-page under https://homieiot.github.io/homie-esp8266/docs/3.0.1/advanced-usage/range-properties/ is outdated, because there is still a "stripNode.advertiseRange" in the codesnippet.
Anyway, my expection is that I have a strip-node and then the properties led_1 till led_x, which should all be settable.

But in my MQTT-broker I see this nodes (for browsing I use the MQTT-Explorer):
```
strip
strip_1
strip_2
strip_3
strip_4
```

And only the strip-node has a property called led (which is settable).
```
strip (node)
  $name = Strip
  $type = strip
  $array = 1-4
  $properties = led
  led (property)
    $name = Led
    $settable = true
    $datatype = boolean
```

And the other nodes strip_1 - strip_4 contains only a $name

For me, the currently handling seems to be wrong and it's maybe a bug-

Thanks :)

nemerdaud, 2024-09-04:
Very old issue, but I have the same problem. Do you received/produced some fix, @thecky ?

nemerdaud, 2024-09-04:
> Very old issue, but I have the same problem. Do you received/produced some fix, @thecky ?
Sorry guys... I was wrong. Actually this behavior is correct based on the 3.X Homie specification. But in the 4.X specification this property type (array) was removed since it's complex to handle and not so featured (based on the discussions here: [https://github.com/homieiot/convention/issues/90](url) ).