homieiot/homie-esp8266 #718: Wemos D1 mini Pro crashes on boot up with debug flag enabled

breezer15, 2021-08-13:
Hello,

when I set build_type = debug, the Wemos crashes on boot up with an exception message.
I use the "awesome-relay" example with platformio.
If I deactivate the debug flag everything works.

```
💡 Firmware awesome-relay (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: f4cfa2d0464a
  • Device ID: pool2
  • Name: Pool
  • Device Stats Interval: 60 sec
  • Wi-Fi:
    ◦ SSID: LAPTOP
    ◦ Password not shown
  • MQTT:
    ◦ Host: 192.30.102.100
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: mqtt
    ◦ Password not shown
  • OTA:
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (2):
epc1=0x3fff4e54 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4e54 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffad4 end: 3ffffdbc offset: 0190
3ffffc64:  3fff4e54 0000049c 007fe000 40106f98  
3ffffc74:  58323025 00000030 00000020 4010082d
3ffffc84:  3fff4e54 0000049c 00000000 0000034f
3ffffc94:  3ffe8958 00001a7c 40100f41 ffffff01  
3ffffca4:  55aa55aa 00000080 00000020 3fff3d3c
3ffffcb4:  00000020 3fff4e54 4010112e 402a1fe2
3ffffcc4:  3fff4e54 00000001 40100a60 402a2e25  
3ffffcd4:  00000000 3fff2194 402a2e71 402276a8
3ffffce4:  00000001 3fff3c00 00000020 3fff3d3c
3ffffcf4:  3fff3c00 3fff11d0 40208fd4 00000001
3ffffd04:  000363ee 4023a6a8 402198e2 00000064  
3ffffd14:  00000001 3fff17d4 4021c1e9 00000002
3ffffd24:  3fff1418 3fff17fc 4021c1e9 00000009
3ffffd34:  3fff462c 3fff13d2 00000020 40243ce0  
3ffffd44:  3fff1130 3ffe99ac 4021692c 40243ce0
3ffffd54:  3fff1130 3fff13d2 40243cf4 40243ce0
3ffffd64:  3fff1130 3ffe99ac 40217265 3fff1130
3ffffd74:  3fff1130 3fff136c 40217288 3fff1130  
3ffffd84:  3fff1130 3fff136c 4020c286 40212eb4
3ffffd94:  00000001 00000013 00000020 3fff3d3c
3ffffda4:  3fff3c00 3fff3c00 4020b85b 00000000  
3ffffdb4:  00000000 00000000 00000000 00000000
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000b4f30
~ld
```

Platformio.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
;lib_deps = Homie
lib_deps = https://github.com/homieiot/homie-esp8266.git#develop
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
build_type = debug
board_build.ldscript = eagle.flash.8m7m.ld
monitor_speed = 115200
```

bertmelis, 2021-08-13:
Reset cause 2 means the ESP has been reset by it's reset-pin.

Try attaching another pin to the "relay".

luebbe, 2021-09-13:
Possibly #713 fixed by #714. Please try the develop branch.

luebbe, 2023-01-16:
Closing due to no activity