homieiot/homie-esp8266 #727: Can't build with arduino-esp32 v2.0.2 (latest stable)

Lucy2003, 2022-02-03:
Hi,

I'm try to compile this library with arduino-esp32 v2.0.2 (latest stable),but the complier threw some error.
Same output on both master and develop branch,just like below:
```
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp: In member function 'virtual void HomieInternals::BootNormal::setup()':
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp:63:139: error: 'SYSTEM_EVENT_STA_GOT_IP' is not a member of 'arduino_event_id_t'
   _wifiGotIpHandler = WiFi.onEvent(std::bind(&BootNormal::_onWifiGotIp, this, std::placeholders::_1, std::placeholders::_2), WiFiEvent_t::SYSTEM_EVENT_STA_GOT_IP);
                                                                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp:64:153: error: 'SYSTEM_EVENT_STA_DISCONNECTED' is not a member of 'arduino_event_id_t'
   _wifiDisconnectedHandler = WiFi.onEvent(std::bind(&BootNormal::_onWifiDisconnected, this, std::placeholders::_1, std::placeholders::_2), WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
                                                                                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp: In member function 'void HomieInternals::BootNormal::_onWifiDisconnected(arduino_event_id_t, arduino_event_info_t)':
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp:367:81: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
   Interface::get().getLogger() << F("✖ Wi-Fi disconnected, reason: ") << info.disconnected.reason << endl;
                                                                                 ^~~~~~~~~~~~
                                                                                 eth_connected
D:\Arduino\portable\sketchbook\libraries\homie-esp8266\src\Homie\Boot\BootNormal.cpp:370:44: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
   Interface::get().event.wifiReason = info.disconnected.reason;
                                            ^~~~~~~~~~~~
                                            eth_connected

```

The `arduino_event_info_t` defination I found is in "esp_event.h":
```
typedef enum {
	ARDUINO_EVENT_WIFI_READY = 0,
	ARDUINO_EVENT_WIFI_SCAN_DONE,
	ARDUINO_EVENT_WIFI_STA_START,
	ARDUINO_EVENT_WIFI_STA_STOP,
	ARDUINO_EVENT_WIFI_STA_CONNECTED,
	ARDUINO_EVENT_WIFI_STA_DISCONNECTED,
	ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE,
	ARDUINO_EVENT_WIFI_STA_GOT_IP,
	ARDUINO_EVENT_WIFI_STA_GOT_IP6,
	ARDUINO_EVENT_WIFI_STA_LOST_IP,
	ARDUINO_EVENT_WIFI_AP_START,
	ARDUINO_EVENT_WIFI_AP_STOP,
	ARDUINO_EVENT_WIFI_AP_STACONNECTED,
	ARDUINO_EVENT_WIFI_AP_STADISCONNECTED,
	ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED,
	ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED,
	ARDUINO_EVENT_WIFI_AP_GOT_IP6,
	ARDUINO_EVENT_WIFI_FTM_REPORT,
	ARDUINO_EVENT_ETH_START,
	ARDUINO_EVENT_ETH_STOP,
	ARDUINO_EVENT_ETH_CONNECTED,
	ARDUINO_EVENT_ETH_DISCONNECTED,
	ARDUINO_EVENT_ETH_GOT_IP,
	ARDUINO_EVENT_ETH_GOT_IP6,
	ARDUINO_EVENT_WPS_ER_SUCCESS,
	ARDUINO_EVENT_WPS_ER_FAILED,
	ARDUINO_EVENT_WPS_ER_TIMEOUT,
	ARDUINO_EVENT_WPS_ER_PIN,
	ARDUINO_EVENT_WPS_ER_PBC_OVERLAP,
	ARDUINO_EVENT_SC_SCAN_DONE,
	ARDUINO_EVENT_SC_FOUND_CHANNEL,
	ARDUINO_EVENT_SC_GOT_SSID_PSWD,
	ARDUINO_EVENT_SC_SEND_ACK_DONE,
	ARDUINO_EVENT_PROV_INIT,
	ARDUINO_EVENT_PROV_DEINIT,
	ARDUINO_EVENT_PROV_START,
	ARDUINO_EVENT_PROV_END,
	ARDUINO_EVENT_PROV_CRED_RECV,
	ARDUINO_EVENT_PROV_CRED_FAIL,
	ARDUINO_EVENT_PROV_CRED_SUCCESS,
	ARDUINO_EVENT_MAX
} arduino_event_id_t;
```

My hardware and software info:
- Arduino IDE 1.8.19 on Windows 7 x64
- arduino-esp32 v2.0.2 with i686 toolchain
- ESP32 Dev Board

Could anyone help me?Thanks.

Lucy2003, 2022-02-03:
Yes,just replace them with new defination,I got it compiled!

luebbe, 2022-02-03:
Glad you got it sorted. Will you close the issue if everything is ok?

closed 2022-02-03 by Lucy2003

rosak, 2024-01-01:
> Yes,just replace them with new defination,I got it compiled!

But which? 'eth_connected' as suggested?  for me, if we're asking for info.disconnected.reason then more logical would be to replace non existing 'disconnected' with 'WIFI_STA_DISCONNECTED' or 'WIFI_AP_STADISCONNECTED' or 'ETH_DISCONNECTED' but I know nothing about it.. @luebbe can you please advise? I am trying to figure out why my WiFi keeps disconnecting every couple of hours on all my ESP32s. Thanks in advance.

luebbe, 2024-01-02:
I honestly don't know, because I never really looked into this issue. Every now and then my ESPs (8266 and 32) lose their WiFi and/or MQTT connection, but they reconnect reliably, so I don't bother. For some devices I had to implement some recovery algorithm so that they resume their previous state properly, but that was it.