homieiot/homie-esp8266 #728: Public variables in PropertyInputHandler are undefined.

mFranz82, 2022-02-25:
Hi!

This is more a question according to custom nodes.

According to this #456 I have another question as well. 

I think it is a scope problem but I am not able to fix this. 

I have a constructor in a class like this with public variables node and PIN. The constructor works as expected but I lose the reference to the PIN number in the hander function :-( 

```

Valve::Valve(const int* PIN, const char* ID, const char* name, HomieNode* node)
: _node(node)
, _pin(PIN)
{
  // Pin has the desired value
  pinMode(*_pin, OUTPUT);

  _node->advertise(ID)
          .setName(name)
          .setDatatype("boolean")
          .settable(  [this](const HomieRange& range, const String& value) -> bool { 
                   // I need the PIN number in the handler function 
                   // But PIN is undefined in this scope. 0
                   printf("Value:  %d\n", _pin);
                   return valveInputHandler(range,value);
          });

bool Valve::valveInputHandler(const HomieRange& range, const String& value) {
  if (value != "true" && value != "false") return false;
  bool on = (value == "true");

  digitalWrite(_pin, on ? HIGH : LOW);

  return true;
}

```

If someone just posts a keyword for a google search that would be absolutely fine. 

- [X] You've read the documentation for *your* release (in the `docs/` folder or at https://homieiot.github.io/homie-esp8266/) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [X] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

luebbe, 2022-02-25:
Maybe take a look at this: https://github.com/luebbe/homie-node-collection? I always subclass my nodes from homie node instead of passing a node around.

Your valve is not different from a relay node. You should be able to use my relay node straight away. At least that's what I'm doing in my valve controlle (with benefits): https://github.com/luebbe/homie-relay-array

mFranz82, 2022-02-25:
Thanks a lot! This is what I need!

closed 2022-02-25 by luebbe